arduino/Arduino #9764: SIGABRT in Arduino IDE 1.8.11 on secondary screen of dual-screen Mac

jtcours, 2020-02-15:
Arduino IDE 1.8.11 sometimes crashes with a SIGABRT. It seems to occur on a Mac with a secondary screen plugged into it, when the IDE tries to open a new window on the secondary screen.

I'm extending the desktop (not mirroring), with macOS Catalina 10.15.3 and Java 1.8.0_102. The system is a MacBook Air with an external monitor, and the crashes seem to happen when the IDE is running on the external monitor.

From /var/log/system.log: 
```
Feb  9 20:19:28 zephyr com.apple.xpc.launchd[1] (cc.arduino.Arduino.18980[12902]): Service exited due to SIGABRT

Feb  9 21:02:39 zephyr com.apple.xpc.launchd[1] (cc.arduino.Arduino.18980[13352]): Service exited due to SIGABRT
Feb  9 21:04:14 zephyr com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x1000001a.Arduino[27021]): Service exited due to SIGABRT
```

From the crash report:
```

Date/Time:             2020-02-09 21:04:14.278 -0800
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        CF37A47B-E000-2C7F-3B88-C469EB1A00D8

Sleep/Wake UUID:       F0468D71-7D75-4004-9946-503F42AE9656

Time Awake Since Boot: 230000 seconds
Time Since Wake:       93000 seconds

System Integrity Protection: enabled

Crashed Thread:        17  Java: Java2D Queue Flusher

Exception Type:        EXC_BAD_INSTRUCTION (SIGABRT)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.

Thread 17 Crashed:: Java: Java2D Queue Flusher
0   libsystem_kernel.dylib        	0x00007fff66f047fa __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff66fc1bc1 pthread_kill + 432
2   libsystem_c.dylib             	0x00007fff66e8ba1c abort + 120
3   libjvm.dylib                  	0x000000010f8482c6 os::abort(bool) (.cold.1) + 125
4   libjvm.dylib                  	0x000000010f664fab os::abort(bool) + 33
5   libjvm.dylib                  	0x000000010f7fe2f9 VMError::report_and_die() + 2301
6   libjvm.dylib                  	0x000000010f259fe3 report_vm_error(char const*, int, char const*, char const*) + 93
7   libjvm.dylib                  	0x000000010f66a9f8 os::get_sender_for_C_frame(frame*) + 76
8   libjvm.dylib                  	0x000000010f25b234 print_native_stack(outputStream*, frame, Thread*, char*, int) + 336
9   libjvm.dylib                  	0x000000010f7fd153 VMError::report(outputStream*) + 2617
10  libjvm.dylib                  	0x000000010f7fddd7 VMError::report_and_die() + 987
11  libjvm.dylib                  	0x000000010f66ae07 JVM_handle_bsd_signal + 802
12  libjvm.dylib                  	0x000000010f66795f signalHandler(int, __siginfo*, void*) + 88
13  libsystem_platform.dylib      	0x00007fff66fb642d _sigtramp + 29
14  ???                           	000000000000000000 0 + 0
15  libawt_lwawt.dylib            	0x0000000115373281 CGLSD_MakeCurrentToScratch + 93
16  libawt_lwawt.dylib            	0x00000001153733d8 OGLSD_MakeOGLContextCurrent + 166
17  libawt_lwawt.dylib            	0x0000000115390e66 OGLContext_SetSurfaces + 188
18  libawt_lwawt.dylib            	0x00000001153956c3 Java_sun_java2d_opengl_OGLRenderQueue_flushBuffer + 1775
19  ???                           	0x0000000119b5ffad 0 + 4726325165
20  ???                           	0x0000000119b382bd 0 + 4726162109
21  ???                           	0x0000000119b382bd 0 + 4726162109
22  ???                           	0x0000000119b382bd 0 + 4726162109
23  ???                           	0x0000000119b30671 0 + 4726130289
24  libjvm.dylib                  	0x000000010f3f3c23 JavaCalls::call_helper(JavaValue*, methodHandle*, JavaCallArguments*, Thread*) + 2243
25  libjvm.dylib                  	0x000000010f3f2462 JavaCalls::call_virtual(JavaValue*, KlassHandle, Symbol*, Symbol*, JavaCallArguments*, Thread*) + 626
26  libjvm.dylib                  	0x000000010f3f26af JavaCalls::call_virtual(JavaValue*, Handle, KlassHandle, Symbol*, Symbol*, Thread*) + 111
27  libjvm.dylib                  	0x000000010f48c2ed thread_entry(JavaThread*, Thread*) + 177
28  libjvm.dylib                  	0x000000010f7a33b3 JavaThread::thread_main_inner() + 373
29  libjvm.dylib                  	0x000000010f7a314f JavaThread::run() + 847
30  libjvm.dylib                  	0x000000010f664937 java_start(Thread*) + 250
31  libsystem_pthread.dylib       	0x00007fff66fc1e65 _pthread_start + 148
32  libsystem_pthread.dylib       	0x00007fff66fbd83b thread_start + 15

Thread 17 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000700008ec2000  rcx: 0x0000700008ec04c8  rdx: 0x0000000000000000
  rdi: 0x000000000000da13  rsi: 0x0000000000000006  rbp: 0x0000700008ec04f0  rsp: 0x0000700008ec04c8
   r8: 0x0000000000000010   r9: 0x0000700008ec0500  r10: 0x0000700008ec2000  r11: 0x0000000000000246
  r12: 0x000000000000da13  r13: 0x0000700008ec0670  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: 0x00007fff66f047fa  rfl: 0x0000000000000246  cr2: 0x000000010f084feb
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133
```

I noticed that OpenJDK has some bug reports of similar behavior in Java2D Queue Flusher but I don't know enough about the inner workings of the Arduino IDE and OpenJDK to know whether they might be related: https://bugs.openjdk.java.net/browse/JDK-8146238 https://bugs.openjdk.java.net/browse/JDK-8203611

jtcours, 2020-02-15:
Similar crash in 1.8.12 when dragging the window from the secondary monitor to the primary.

From /var/log/system.log:
```
Feb 14 21:12:49 zephyr com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000025.Arduino[41629]): Service exited due to SIGABRT
```

Gory details attached:
[arduino_1.8.12_crash.txt](https://github.com/arduino/Arduino/files/4207779/arduino_1.8.12_crash.txt)

per1234, 2020-02-15:
Here's another similar report:
https://forum.arduino.cc/index.php?topic=661342.msg4466997#msg4466997
>after I updated the IDE to 1.8.11
my IDE is crashing when I want to use it on a secondary Monitor.
Im on OSX 10.15.2.

lathoub, 2020-02-15:
Same for 1.8.12

dcolling, 2020-02-28:
I've got the same issue, but it is crashing on either screen and not necessarily when I'm moving the window, however, that is the most common occurrence.

hussamaldean, 2020-02-29:
mine crashes when macbook pro changes gpu

dirkx-reno-2020, 2020-04-30:
It can be circumvented / workaround by ticking the somewhat obscure -'Displays have separate Spaces' options in System Preferences -> Mission control.

morrissey007, 2020-05-04:
> It can be circumvented / workaround by ticking the somewhat obscure -'Displays have separate Spaces' options in System Preferences -> Mission control.

Thank you but this makes no difference for me, it crashes regardless of this setting.

dgorbunov, 2020-05-17:
Experiencing the same issue in Catalina, I've set my main display to my monitor so I don't have to drag the IDE over when it starts, causing it to crash.

woutervdijke, 2020-05-18:
This crash seems to happen on any change to the displays: connecting an external display, disconnecting it, moving the Arduino IDE to a different display.

a7hybnj2, 2020-07-07:
This effects me too. If I leave the Arduino IDE open when my computer sleeps when it wakes up crash. Same circumstance.

cmaglie, 2020-07-07:
This bug has been solved with the latest release 1.8.13. It looks like we forget to close this issue among the other duplicates.

closed 2020-07-07 by cmaglie