arduino/Arduino #9768: hooks.savehex.postsavehex is not really called

yahyatawil, 2020-02-16:
I noticed that hex-related hooks are not really called. 

IDE version: 1.8.4 & 1.8.13
OS: Windows 10
Details:

I added the following to platform.txt 

`recipe.hooks.sketch.prebuild.1.pattern = {build.source.path}\fw_ver.bat GIT {build.path} {build.source.path}
recipe.hooks.savehex.postsavehex.2.pattern = {build.source.path}\fw_ver.bat SAVE {build.path} {build.source.path}`

only the first hook is called not the second one. I tried to change the pattern number with no success.

matthijskooijman, 2020-02-16:
Do you also have a `recipe.hooks.savehex.postsavehex.1.pattern`? AFAIU the numbers are used to distinguish multiple hooks of the same type, so they should start at 1 for each hook type (IIRC this changed a while ago, before the numbers could be anything, but to fix some other problem the code was changed to only start at 0 or 1 and go upwards, rather than enumerating all of them)

yahyatawil, 2020-02-16:
No I have only the mentioned hooks.

matthijskooijman, 2020-02-16:
There was a suggestion hidden in my comment. Just in case it was unclear: Could you change the 2 to a 1 in your pattern and see if that does work?

yahyatawil, 2020-02-16:
Sure. 

No difference. 
Also tried to make the hook only echo a text, same ... not working. 

This is the complete verbose:

```
C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\DELL\AppData\Local\Arduino15\packages -hardware C:\Users\DELL\Documents\Arduino\hardware -tools C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\DELL\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\DELL\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path C:\Users\DELL\AppData\Local\Temp\arduino_build_704916 -warnings=none -build-cache C:\Users\DELL\AppData\Local\Temp\arduino_cache_161131 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -verbose C:\Users\DELL\Desktop\firmware_versioning\fw\fw.ino
C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\DELL\AppData\Local\Arduino15\packages -hardware C:\Users\DELL\Documents\Arduino\hardware -tools C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\DELL\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\DELL\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10813 -build-path C:\Users\DELL\AppData\Local\Temp\arduino_build_704916 -warnings=none -build-cache C:\Users\DELL\AppData\Local\Temp\arduino_cache_161131 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\DELL\Desktop\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -verbose C:\Users\DELL\Desktop\firmware_versioning\fw\fw.ino
Using board 'mega' from platform in folder: C:\Users\DELL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.200
Using core 'arduino' from platform in folder: C:\Users\DELL\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.200
Detecting libraries used...
"C:\\Users\\DELL\\Desktop\\arduino-nightly-windows\\arduino-nightly\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\DELL\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.200\\cores\\arduino" "-IC:\\Users\\DELL\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.200\\variants\\mega" "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916\\sketch\\fw.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\DELL\\Desktop\\arduino-nightly-windows\\arduino-nightly\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\DELL\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.200\\cores\\arduino" "-IC:\\Users\\DELL\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.6.200\\variants\\mega" "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916\\sketch\\fw.ino.cpp" -o "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\DELL\\Desktop\\arduino-nightly-windows\\arduino-nightly\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916\\preproc\\ctags_targe [... cut ...]
"C:\\Users\\DELL\\Desktop\\arduino-nightly-windows\\arduino-nightly\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916/fw.ino.elf" "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916/fw.ino.hex"
"C:\\Users\\DELL\\Desktop\\arduino-nightly-windows\\arduino-nightly\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\DELL\\AppData\\Local\\Temp\\arduino_build_704916/fw.ino.elf"
Sketch uses 3430 bytes (1%) of program storage space. Maximum is 253952 bytes.
Global variables use 278 bytes (3%) of dynamic memory, leaving 7914 bytes for local variables. Maximum is 8192 bytes.
```

per1234, 2020-02-16:
Are you doing a **Sketch > Export compiled Binary**? The `savehex` hooks are only called when you do that, because the `savehex` recipe is only run when you do that. The `savehex` hooks are not called during a regular "Verify/Compile".

yahyatawil, 2020-02-17:
This explains! I didn't use the export option. 

I was little confused as the hooks in the documentation are under this title "Pre and post build hooks". 

I think it is better to add hooks for both cases or at least make it clear in the [documentation ](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification)

closed 2020-02-19 by yahyatawil