arduino/Arduino #9769: My updated library not found by library manager

ramonheras, 2020-02-16:
Hi,

Yesterday I updated the library ( https://github.com/ramonheras/Pixel-and-Play-Arduino-Library ) up to the version 1.1.0 but library manager just finds prior versions. It may happens because I changed the name and description alongside the version.

per1234, 2020-02-16:
It's no problem to change the description, but the Library Manager index does need to be updated after you change the `name` value, which we are happy to do upon request. Unfortunately, the new `name` value is not compliant with the requirements for inclusion in the Library Manager index:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces ( ), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number. Note that libraries with a name value starting with Arduino will no longer be allowed addition to the Library Manager index as these names are now reserved for official Arduino libraries.

Your current `name` value contains the disallowed characters `(` and `)`. Please change the name to be compliant and then comment here once that is done.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ramonheras, 2020-02-16:
Thanks, changed!

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie