arduino/Arduino #9772: [Library manager] Please add Arduino-HomeKit-ESP8266 to the library manager index

Mixiaoxiao, 2020-02-16:
Please add Arduino-HomeKit-ESP8266 to the library manager index 

[https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266)

per1234, 2020-02-16:
Hi @Mixiaoxiao. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Mixiaoxiao, 2020-02-17:
Hi! 
Thanks for your response.
I have changed the `name` of this library (without starting with the word "Arduino")

cmaglie, 2020-02-17:
Done.

closed 2020-02-17 by cmaglie