arduino/Arduino #9773: Checksums for 1.8.12 are md5, not sha512

JChristensen, 2020-02-17:
On the [Download page](https://www.arduino.cc/en/Main/Software), the [Checksums link](https://downloads.arduino.cc/arduino-1.8.12.md5sum.txt) contains MD5 checksums, not SHA512 as stated on the page, and as has been used for several release now.

MD5 is severely compromised and should not be used. Users should not install 1.8.12 unless they can validate the download with a stronger checksum.

facchinm, 2020-02-17:
It's not perfectly clear and maybe it needs a rewording, but clicking on `Checksums` brings you to the legacy md5, while clicking on `(sha256)` holds the sha256 checksums.

closed 2020-02-17 by facchinm

JChristensen, 2020-02-17:
My bad, sorry about that. I might just do away with the MD5 checksums.