arduino/Arduino #9781: Problems with 2nd monitor using Arduino IDE

Kauzge, 2020-02-18:
I experienced a bug with the Arduino IDE when using a second monitor:

1. I plugged in my 2nd monitor
2. opened the Arduino IDE
3. moved the Adruino IDE window to my 2nd screen
4. (after work on sketch was finished) closed the window on 2nd screen

A few days later I tried to open the same sketch, but this time with no 2nd monitor connected.
The Arduino IDE would only show an icon in the taskbar. The program would not open, and just show a blank preview if I hover over the taskbar-icon.

My solution for the problem was to plug in the 2nd screen. The IDE window showed up there. I had to move it to the main screen and close it there to prevent the bug from appearing again.
I'm not used to this behaviour, so it took quite a long time to figure out what could be the cause of the problem. Other programs usually open on the first monitor if no 2nd monitor is connected, even if they got closed on the 2nd screen.

My system:
Microsoft Surface Pro
Windows 10
Arduino IDE 1.8.12 (Installer)

facchinm, 2020-02-19:
Hi @Kauzge ,
the issue is probably related with https://github.com/arduino/Arduino/issues/5581 and should be solved by https://github.com/arduino/Arduino/pull/6589 (maybe only for the serial monitor, not the IDE itself, taking a closer look later today)

Kauzge, 2020-02-19:
Hello @facchinm ,
this is exactly my problem. I'm sorry that I opened another issue on an existing topic. Before I opened the issue, I searched for existing reports, but couldn't find any. 
Should I close this issue?

sushantag9, 2020-02-26:
> Should I close this issue?

yes it would be great if your problem is resolved

facchinm, 2020-02-28:
Hi @Kauzge ,
I submitted https://github.com/arduino/Arduino/pull/9814 to resolve your problem. In a bit a build should be produced; if you could test it and report the results it would be great.

closed 2020-03-23 by cmaglie

catalinroweb, 2020-04-23:
The problem is still here on MacOS Catalina 10.15.4 (also on older versions of Catalina).

ondr3j, 2020-07-13:
Still happening on Windows 10 as well. Closed IDE and disconnected secondary monitor. If I reopen IDE, the loading screen shows up and that's that. IDE window for last sketch is "invisible".

PR #9814 (2020-03-23): Don't restore sketch location if last.sketch.location is out of display
`if ((screen.width != screenW) || (screen.height != screenH))` in Base.java was making sure to reset every preference in case the display geometry changed.
Unfortunately, on Windows 10, `Toolkit.getDefaultToolkit().getScreenSize()` reports only the primary monitor either if external display is connected or not.

Fixes #9781

### All Submissions:

* [x] Have you followed the guidelines in our Contributing document?
* [x] Have you checked to ensure there aren't other open [Pull Requests](https://github.com/arduino/Arduino/pu [... cut]
commit 558ddd3d (2020-03-23): Don't restore sketch if last.sketch.location is out of display

`if ((screen.width != screenW) || (screen.height != screenH))` was making sure to reset every preference in case the display geometry changed.
Unfortunately, on Windows 10, `Toolkit.getDefaultToolkit().getScreenSize()` reports only the primary monitor either if external display is connected or not.

Fixes #9781