arduino/Arduino #9782: ARDUINO MKR 1400 is Slow / Unreliable when using the SMS.available() function

Kevin-a69, 2020-02-18:
I am currently experiencing a significant performance issue on the Arduino MKR 1400 regarding its handling of SMS incoming messages using the "available()" function.

For some reason there are major delays in receiving the SMS messages whenever multiple SMS are sent to the MKR 1400

I am using a slightly modified version of the ReceiveSMS (example sketch) to demonstrate the issue.
The original example setup()  is unchanged

Here is the modified main loop :

      void loop() {

      if (sms.available())  {

           sms.peek();
       
           while (! sms.ready()) {
           delay(100);
           }
      
           sms.flush();

           while (! sms.ready()) {
           delay(100);
           }

           Serial.println("MESSAGE RECEIVED");
           }
       }

So for example I would send 5 short SMS to the MKR 1400 (each one right after the other )

Then on my PC's serial monitor I would see the first message come in and then some seconds after the 2nd SMS would show up and then several seconds after the 3rd would show and then sometimes the other 2 messages take quite long or sometimes aren't even received.

AND if I take that same telephone SIM and place it in a cellphone and I repeat the same 5 SMS tests ...... ALL 5 SMS would be instantly received on the cellphone without ANY delay.  
Therefore it is definitely not an issue with the GSM service provider.

I just can't figure out what could be causing such large delays on the MRK 1400 board when handling the incoming SMS but it is a real show stopper for the application I am developing.
AND considering that I purchased this board specifically for working with incoming/outgoing SMS

I have not tested to see if there are also problems sending out SMS
At this time my first hurdle is getting this MRK 1400 board to instantly and reliably function using the SMS.available() function OR SOME OTHER WAY.

I could not find any solution for this problem within the https://forum.arduino.cc  (recommended by Arduino)
I have posted several times there in the past weeks and have gotten no response on this matter.

Therefore I am posting the issue here HOPING that I can get some EXPERT assistance/guidance on this matter.

janakelarsson, 2020-02-27:
I had a look at the MKR GSM 1400 unit but decided for a MKR NB 1500 unit instead. 3G is still available here (I think) but not for long, the operators are switching to 4G. People are complaining that the MKR GSM only connects 2G, and then delays of the kind you describe are not strange.

SMS itself is not all that reliable, you are not even guaranteed delivery.

Kevin-a69, 2020-02-28:
Thanks for your response Mr. Larsson

If this is so then it is indeed disappointing to say the least
Because I thought that the SMS protocols would be the most simple and reliable procedure to communicate via the cellular system.
Are you saying that the SMS low reliabililtiy is charactersitic of the MKRGSM boards or due to bugs in the MKRGSM Library functions ?
AND while we are on this topic may I also ask your opinion on which other options (within or outside the Arduino technologies) would be worth the while considering for developing a project that requires the use of SMS communication and of course basic automation.

Thanks for your responses.

janakelarsson, 2020-02-28:
SMS reliability has nothing to do with the MKR GSM boards. SMS reliability is usually good, but there is no guarantee that SMS is even delivered, for any carrier, for any phone. GSM/3G/4G is simply not built that way.

There are some bugs in the library functions, but I do not think they would affect you. (Unless you send the string "OK" as an SMS, that might actually give problems, I have submitted a patch for that problem.)

You should check if your modem connects 2G or 3G. There is example code in the library for that. If you connect 2G only (which is possible, not all carriers have 3G anymore, your phone probably uses 4G only), then I have read elsewhere that power delivery to the board may be a problem, and may cause stability issues.

I have built a simple monitoring system using a MKR NB 1500 board and some sensors. This works fine and is reliable. It receives commands via SMS, and responds via SMS. It also sends SMS when reaching specific thresholds for the sensors. I have built it to restart the network connection on errors, but it seems to be very stable.

Kevin-a69, 2020-03-03:
Hello again my friend,
I have another little challenge I could use some help with.

When I  communicate directly with the SARA U201 (the communication chip within the MKR1400 board) using the UBlox application along with the "SerialPassthrough" Ardunio example sketch, I am able to issue the command " AT+CMGD=4 " which deletes ALL the messages from the SARA's message storage; including unread messages.
The technical guys from the Arduino Store tells me that there is currently no such Arduino Function within the MKRGSM Library that I could use to achieve this result.  And they offer no further help to me on this matter ..... except to say "try the forums"

What I need is to execute that same SARA command (  AT+CMGD=4  ) BUT from within a "running" Arduino Sketch instead of manually via the UBlox external tool.

I figure there must be a way to send AT+ commands to the onboard SARA U201 chip via a sketch

Any suggestions ?
Thanks again.

Thank you.

janakelarsson, 2020-03-03:
As far as I understand, "AT+CMGD=4" deletes message number 4, while "AT+CMGD=,4" would delete all messages (note the extra comma). I don't have a unit here to test.

Check the source code for the MKRGSM library, in GSM_SMS.cpp there is a "flush()" method, you could use that as a template for a "clear()" method. Or just for using it in a sketch.

janakelarsson, 2020-07-23:
@Kevin-a69 I have now tested on a different modem and you will need to issue "AT+CMGD=0,4" to delete all messages. See https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/MKRGSM/issues/128

closed 2021-03-28 by per1234