arduino/Arduino #9783: TFMini-Plus-I2C Arduino  Library

budryerson, 2020-02-19:
Hi folks,

I am so sorry to have to bother you with this.

A version 0.4.0 of the TFMini-Plus-I2C library shows up in the Arduino Library Manager.
This is wrong.  There is no version 0.4.0.  I don't know how this happened
and I do not know how to fix this, or how to make it go away.

The correct and current version is v0.2.2.
Please help me to make  v0.4.0 go away.
Can you fix this?  Please?

Again, my apologies.

Bud Ryerson
San Francisco

per1234, 2020-02-19:
The reason is the `version` field in library.properties was set to `0.4.0` when the `0.1.0` tag was created:
https://github.com/budryerson/TFMini-Plus-I2C/blob/0.1.0/library.properties#L2

You can fix this by deleting the `0.1.0` tag:
https://github.com/budryerson/TFMini-Plus-I2C/releases/tag/0.1.0
deleted tags are processed by the indexer job every Sunday, so the 0.4.0 version will still be available in Library Manager until then.

closed 2020-02-19 by per1234

budryerson, 2020-04-26:
Hello again,

The `0.1.0` tag is gone (I believe), and many Sundays have passed since then.
The Arduino library is still showing version `0.4.0` as the latest version.
That is not correct.  Version `0.2.2` is the latest,
I could use your advice, but I would prefer your assistance because in this instance it appears that the indexer is not doing its job correctly.

Thank you very much,

Bud Ryerson
San Francsico

per1234, 2020-04-27:
We'll use your new issue https://github.com/arduino/Arduino/issues/10100 for tracking this and any further discussion.