arduino/Arduino #9787: Add an `exclude` field in library.properties specification // allows files exclusion for library build

MonsieurV, 2020-02-20:
Hi,

# Context: library with "extra" content

I'm using the [library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) to publish [a library](https://github.com/MonsieurV/ArduinoPocketGeiger) to the [Arduino Library](https://www.arduinolibraries.info).
I've recently added [a build log section](https://github.com/MonsieurV/ArduinoPocketGeiger/tree/master/buildlogs) to my library, as part of the documentation.

# Problem: extra content included in library zip, making it bulkier

These build logs contain many images, which are getting included with the rest of the library to the directory. The result is a library zip size inflation:
[image]

(See [library profile](https://www.arduinolibraries.info/libraries/radiation-watch))

This build logs directory was not intended to be published on the library directory / added to the final library zip.

# Suggestion: add a mechanism for specifying files to be excluded from the library zip

Add an `exclude` field in the `library.properties` specification, allowing library publisher to define a list of files/files pattern to be excluded from the final library build. This would allows publisher to add extra content to their GitHub repositories without it to be systematically published along the library on the Arduino library directory.

Related: #7315 #5688

bengtmartensson, 2020-02-22:
Really useful feature -- how did we get by without it?

However, `library.properties` is a file that tells _the user and his tools_ how to handle the library; information on how the library manager is extracting stuff from Git does not belong there. Possibly a new file?

Additionally, a positive version (i.e. `include` instead of `exclude`) may be more useful. Of course, offering both possibilities would be optimal.

per1234, 2021-04-02:
Thanks to bengtmatensson for an explanation of how `library.properties` is not an ideal mechanism for this. I think the alternative proposal of using the standardized functionality already provided by Git will be a better approach so I'll close this in favor of that one: https://github.com/arduino/Arduino/issues/8478

closed 2021-04-02 by per1234