arduino/Arduino #9788: Nano EVERY mis-identified

BallscrewBob, 2020-02-20:
In IDE 1.8.12 there appears to be an issue with the nano every being categorised wrongly.

Global variables use 31 bytes (0%) of dynamic memory, leaving 6113 bytes for local variables. Maximum is 6144 bytes.
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description

[image]

CREATE editor does not have the issue so it seems to be confined to desktop IDE's

BallscrewBob, 2020-02-20:
Update.
Also appears in 1.8.11
Recently updated all the cores so maybe it is a core issue ?

facchinm, 2020-02-20:
Hi @BallscrewBob,
the part after the comma is the "additional menu" config (and it means the core is able to compile some code with ATMega328 compatibility).
You can switch back to 4809 "plain" under Tools menu

closed 2020-02-20 by facchinm

BallscrewBob, 2020-02-20:
Just saw the 4809 option.
Switched  but same issues so guess tone examples wont run on the every.