arduino/Arduino #9792: Add library to the arduino library manager

mateusjunges, 2020-02-21:
Good night! 

As described [here][docs], i'm opening this issue to add my new library to the library manager.

Can you please add it to the arduino library manager?

[Link to download it][link-1]: https://github.com/mateusjunges/accel-stepper-with-distances 

Thanks in advance!

[link-1]: https://github.com/mateusjunges/accel-stepper-with-distances
[docs]: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

mateusjunges, 2020-03-02:
@per1234 any updates on this?

per1234, 2020-03-02:
Hi @mateusjunges. Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

mateusjunges, 2020-03-02:
@per1234 Thanks for your answer. I'll wait for this.

cmaglie, 2020-03-02:
Done. Sorry for the long delay.

closed 2020-03-02 by cmaglie

mateusjunges, 2020-03-02:
@cmaglie Thanks! 😊