arduino/Arduino #9794: Wire.begin() doesn't initialize the SDA 20 and SCL 21 of arduino due

rafaccount123, 2020-02-21:
Hello,

I'm using the Wire library.

I try to communicate between the arduino due and the VEML6030 of sparkfun via the pins SDA 20 and SCL 21. But when I write Wire.begin() and try to communicate in I2C via SDA 20 and SCL 21 between the VEML6030 and the arduino due I can't. The connections are fine and the VEML6030 and the arduino due are well powered.

Maybe there is an issue in the library Wire for the arduino due to use SDA 20 and SCL 21.

Thanks for the help.

dsyleixa, 2020-02-23:
for my Due i2c actually worked (last tested: IDE 1.8.6 , test fot 1.8.9 still pending) - so:
a) which is your Arduino IDE version?
b) what does a i2c scanner show if you attach your device?
c) how does it work with different i2c devices

dsyleixa, 2020-02-24:
just tested anew:
I2C at pins 20+21 works like a charm!
IDE 1.8.9
Due board 1.6.12

rafaccount123, 2020-03-10:
Thanks for the answer.

I didn't try i2c scanner program with differents i2c test devices.

The following code isn't working well:

`
#include <VEML6030.h>

VEML6030 ALS;

void setup() {
  Serial.begin(38400);  //Begin Serial
  ALS.begin();  //Begin the UV module
  

}

void loop() {
  ALS.AutoRange(); //Determine the best gain and integration time values for given light intensity
  delay(1000); //Wait for new value to be taken, up to 1 second
  Serial.print("Lux = ");
  Serial.print(ALS.GetLux());  //Get compensated UVA value
  Serial.print(" White = ");
  Serial.print(ALS.GetWhite());  //Get compensated UVB value
  Serial.print(" Ambient Raw = ");
  Serial.println(ALS.GetALS());
  delay(1000);
}
`
The link of  the code is [https://github.com/NorthernWidget/VEML6030/blob/master/Examples/VEML6030_Demo/VEML6030_Demo.ino](url)

I get Lux=494780224 White=494780224 Ambient Raw=494780224 with the VEML6030 powered or not powered. I see the results with serial.print.

The transmission of data isn't working with i2c at pins 20+21.

The pcb which is the support of the VEML6030 is made by myself.

I took my inspiration on the datasheet of the VEML6030 with the following schematic
[image]

The ADDR and the GND of the datasheet are joined together on pcb.

So when I write in the code `ALS.begin(0x10)` the Lux=White=Ambient Raw=0.

My IDE arduino is 1.8.12 and arduino due version I don't know
How you know the version of the due board?

In conclusion I don't understand why the lux value doesn't change following the light.

Thanks for the help.

rafaccount123, 2020-03-10:
Thanks for the answer.

I commented today about the issue in github
This issue remains.

 

M. Madeira Rafael
CIFEC - Technicien electronique 
12 bis rue du Cdt Pilot - 92200 Neuilly sur Seine - FRANCE 

Tel direct: +33 1 4640 4944 / Tel: +33 1 4640 4949
Site internet : www.cifec.fr 
Boutique : www.shop.cifec.fr

dsyleixa, 2020-03-10:
> How you know the version of the due board?

Tools -> Board -> Boardmanager: search filed "Due":
lists: Arduino SAM Boards 

as to IDE 1.8.12, in case perhaps this issue is new to this version, but you'd better test both
-  different i2c devices on the Due, 
- your VEML6030 by a i2c scanner program on the Due, and
- your VEML6030 at a different Arduino (e.g., a Nano, a ESP32, or a SAMD21).

Which i2c port detects your Due with this scanner program when you attach your VEML6030 ?

```
 // i2c_scanner
 //    
 // This sketch tests the standard 7-bit addresses
 // Devices with higher bit address might not be seen properly.
 //
     
 #include <Wire.h>
    
 byte error, address;
 int nDevices, nDeverr;
 
 void setup()
 {            
      //Wire.pins(D20, D21);   // only for test reasons, should work even outcommented
      Wire.begin();
 
      Serial.begin(115200);
      while (!Serial);                 // Leonardo: wait for serial monitor

      Serial.println("\nI2C Scanner");     
                
 }
     
     
 void loop()  {    
      static uint32_t  i2cclock = 10000; 
      int iores;
      
      nDevices = 0;
      nDeverr  = 0;
      
      if(i2cclock==10000) i2cclock=100000; 
      else 
      if(i2cclock==100000) i2cclock=400000;
      /*else 
      if(i2cclock==400000) i2cclock=1000000;
      */
      else 
      if(i2cclock==400000) i2cclock=10000;
      
      
      Wire.setClock(i2cclock);
      Serial.print("\nScanning at "); Serial.print(i2cclock/1000); Serial.println("k");
      
      for(address = 0; address < 128; address++ )   {

        if (address%16 == 0)  {   
          Serial.println();  
          Serial.print( (address+1)/16);
          Serial.print("  ");
        }

        
        if(address==0 || address==127) {
           Serial.print("** ");
           continue;
        }
        
        Wire.beginTransmission(address);
        error = Wire.endTransmission();
     
        if (error == 0)
        {          
          if (address<16) Serial.print("0"); 
          Serial.print(address,HEX); Serial.print(" ");         
          nDevices++;
        }
        else if (error==1)
        {
          Serial.print("E1 ");   
          nDeverr++;
        }         
        else if (error==3)
        {
          Serial.print("E3 ");   
          nDeverr++;
        } 
        else if (error==4)
        {
          Serial.print("E4 ");   
          nDeverr++;
        }    
        else // error==2 == no device
        {      
           Serial.print("-- ");   
        }          
      }
      
      Serial.println();
      Serial.print("Scan result for i2cclock "); Serial.print(i2cclock/1000); Serial.println("k");
      Serial.print("found: "); Serial.print(nDevices); Serial.println(" devices ");
      Serial.print("error: "); Serial.print(nDeverr ); Serial.println(" devices \n");
      delay(5000);      
 }

```

rafaccount123, 2020-03-16:
Thanks,

My Due board is 1.6.12.  I'm now using the sparkfun ambient light sensor - VEML6030 ( SEN-15436) which costs 4.95$. [https://www.sparkfun.com/products/15436?_ga=2.26399741.1939773733.1584350078-1168471852.1567063892](url)

I used the scanner program and the results are as follows:

[image]

[image]

[image]

What am I to conclude ?
Thanks

dsyleixa, 2020-03-16:
there is a device detected at hex 0x48, which means: appearenty  i2c works well.
If i2c didn't work then it wouldn't detect any devices at all.

dsyleixa, 2020-03-20:
@rafaccount123:
 I would actually appreciate if you reacted faster to the advices, not always almost an entire week pause in between - 
or close the issue if it's resolved

rafaccount123, 2020-03-20:
The issue isn't resolve because the lux value of the VEML6030 is always equal to 0.
I follow your advice but I can't find the driver lib issue or the source code issue.

dsyleixa, 2020-03-20:
can you contact the authors of the driver lib or the example code?
can you test another i2c device, e.g. a OLED display, a PCF8574, a MCP23017, a MPU6050  or whatever i2c duevice you might have, just to see if it works?

rafaccount123, 2020-03-20:
I have a rgb lcd shield 16*2 blue/white display of adafruit [https://www.adafruit.com/product/714](url) 
The display works like a charm on the SCL1 and SDA1 but doesn't work on SDA20 and SCL21 because the code is made to work  only on the SCL1 and SDA1. I don't have others i2c devices.
I can contact the authors of the driver lib VEML6030 and the example code but I didn't find the issue for the moment.

dsyleixa, 2020-03-20:
I s this the lib you are using?
https://github.com/NorthernWidget/VEML6030/blob/master/VEML6030.h

perhaps just open a new issue over there?

rafaccount123, 2020-03-20:
yes, is this lib exactly. I will open a new issue over there .

rafaccount123, 2020-03-30:
Hello, the issue is always there.

I installed Arduino SAM Boards (32-bits ARM Cortex-M3) version 1.6.12 which is arduino due in 
Tools -> Board -> Boardmanager a long time ago. I selected arduino due (programming port)

But when I try to locate the path of the library Wire of the arduino due I can't find.
I just find the path of the library Wire of the arduino uno which is for me the following 
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 

I find the path of the avr cards and don't find the path of arm cards.

How to find the path of the arm cards and be sure that the Wire library used is the good library used to arduino due ?

Thanks in advance.

dsyleixa, 2020-03-30:
the wire() lib is located automatically when using either board and compiling to that core, you don't have to worry.
My Arduino Due works like a charm with Wire pins 20+21, and your Due too, because your Due detects your device fine by the i2c scanner.

rafaccount123, 2020-03-31:
Hello, I tried to locate \packages\arduino\hardware\sam\1.6.12\libraries\Wire but I can't find it.
Can you write the complete path to see if I locate it, please ?

per1234, 2020-03-31:
Do this:

1. (In the Arduino IDE) **Tools > Board > Arduino Due (Programming Port)**
2. **File > Examples > Wire > digital_potentiometer**
3. **Sketch > Show Sketch Folder**

This will open the `libraries/Wire/examples/digital_potentiometer` subfolder of the Arduino SAM Boards platform installation folder. The Wire library is at `libraries/Wire`.

rafaccount123, 2020-03-31:
Thanks, I found the correct library in C:\Users\Antonio\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\Wire\examples\digital_potentiometer

The library wire of arduino due is there.

rafaccount123, 2020-04-01:
Hello, I used a new sensor named BH1750FVI and the a new library BH1750FVI located here : https://github.com/PeterEmbedded/BH1750FVI . The code is the following:

`
/*
  This example uses only 4 wires to test the BH1750 Light sensor
  
  Connecting the sensor to a NodeMCU ESP8266:
  VCC  <-> 3V3
  GND  <-> GND
  SDA  <-> D2
  SCL  <-> D1
  Connecting the sensor to a Arduino UNO:
  VCC  <-> 3V3
  GND  <-> GND
  SDA  <-> A4/SDA 
  SCL  <-> A5/SCL
  Connecting the sensor to a Arduino DUE:
  VCC  <-> 3V3
  GND  <-> GND
  SDA  <-> D20/SDA
  SCL  <-> D21/SCL
*/

#include <BH1750FVI.h>

// Create the Lightsensor instance
BH1750FVI LightSensor(BH1750FVI::k_DevModeContLowRes);

void setup() 
{
  Serial.begin(115200);
  LightSensor.begin();  
}

void loop()
{
  uint16_t lux = LightSensor.GetLightIntensity();
  Serial.print("Light: ");
  Serial.println(lux);
  delay(250);
}
`

This code is inteh example of the library.

The results are the following in lux:

Light: 40
Light: 36
Light: 40
Light: 40
Light: 13
Light: 3
Light: 0
Light: 0

The sensor works correctly. conclusion is the library of the VEML6030 who doesn't works correctly.

The only issue is that we need to make extra work on the code to use the SCL1 and SDA1 because the Wire.being() doesn't initiate automaticaly  the two I2C ports. (SCL1, SDA1 and SDA20, SCL21)

per1234, 2020-04-01:
>conclusion is the library of the VEML6030 who doesn't works correctly.

Thanks for the update @rafaccount123. Since you have concluded that your problem is not caused by a defect in official Arduino software, I'm going to close this. If further investigation does identify a bug in Arduino's software, you can comment here and I'll re-open the issue.

closed 2020-04-01 by per1234

fubar-life123, 2020-06-06:
> 
> 
> Hello, I used a new sensor named BH1750FVI and the a new library BH1750FVI located here : https://github.com/PeterEmbedded/BH1750FVI . The code is the following:
> 
> ```c++
> /*
>   This example uses only 4 wires to test the BH1750 Light sensor
>   
>   Connecting the sensor to a NodeMCU ESP8266:
>   VCC  <-> 3V3
>   GND  <-> GND
>   SDA  <-> D2
>   SCL  <-> D1
>   Connecting the sensor to a Arduino UNO:
>   VCC  <-> 3V3
>   GND  <-> GND
>   SDA  <-> A4/SDA 
>   SCL  <-> A5/SCL
>   Connecting the sensor to a Arduino DUE:
>   VCC  <-> 3V3
>   GND  <-> GND
>   SDA  <-> D20/SDA
>   SCL  <-> D21/SCL
> */
> 
> #include <BH1750FVI.h>
> 
> // Create the Lightsensor instance
> BH1750FVI LightSensor(BH1750FVI::k_DevModeContLowRes);
> 
> void setup() 
> {
>   Serial.begin(115200);
>   LightSensor.begin();  
> }
> 
> void loop()
> {
>   uint16_t lux = LightSensor.GetLightIntensity();
>   Serial.print("Light: ");
>   Serial.println(lux);
>   delay(250);
> }
> ```
> 
> This code is in the example of the library.
> 
> The results are the following in lux:
> 
> Light: 40
> Light: 36
> Light: 40
> Light: 40
> Light: 13
> Light: 3
> Light: 0
> Light: 0
> 
> The sensor works correctly. conclusion is the library of the VEML6030 who doesn't works correctly.
> 
> The only issue is that **we need to make extra work on the code to use the SCL1 and SDA1 because the Wire.being() doesn't initiate automaticaly the two I2C ports.** (SCL1, SDA1 and SDA20, SCL21)

There is a problem with the Arduino Library!???

fubar-life123, 2020-06-06:
I tested my I2C sensors on a raspberry PI, they were recognized immediately! A problem initializing the I2C ports is definitely a problem with the Arduino Wire Library! I can't even find an address using I2C multi-speed Scanner (latest version). It's either board termination (1k resistors which are too low) or the issue with "wire.begin" or a combination of both! Users are not reporting issues on AVR Arduino boards.

dsyleixa, 2020-06-07:
I never run into issues with my Due ecept the libs are using proprietary AVR code (which should be depricated fo every arduino  lib!) or the Due was intended to be used as an i2c slave via 20+21=i2c0 (via i2c1 then it worked though).
Just to mention, the Due has 2 seperate built-in  i2c pullup sets: 1 on-board, plus 1 on-chip, both are weak, but one may always apply always additionally external ones (e.g., 3k3 each).
I have also a Raspi, and I never found i2c devices which work with the Raspi but don't with the Due.

But I agree, the Due is very badly maintained by the Arduino devs.

dsyleixa, 2020-06-07:
I never run into issues with my Due  - except the libs are using proprietary AVR code (which should be deprecated fo either arduino  lib!) or the Due was intended to be used as an i2c slave via 20+21=i2c0 (via i2c1 then it often worked though because there are no built-in pullups).
One may even remove the Due pullups if one wishes: https://forum.arduino.cc/index.php?action=dlattach;topic=223513.0;attach=72003
I have also a Raspi, and I never found i2c devices which work with the Raspi but don't with the Due.

But I agree, the Due is very badly maintained by the Arduino devs.

fubar-life123, 2020-06-07:
So just to update, I purchased a Mega2560 Clone. Since it is an AVR
microcontroller, I2C wire library should work. Unfortunately, I'm still not
able to see the device using the I2C scanner... Since it did work with the
Raspberry Pi and not with (2) Arduino boards, it is safe to say that it is
indeed a *termination issue* (in my case). I may resolve the issue of not
being able to find the device, but run into another issue with ack nack on
the Due. But not being able to see the device at all is probably a
termination problem. I think it's best practice *not to preemptively
terminate traces* if there is potential for incompatibility with specific
devices. Resistors are cheap. Anyone can stick them on a breadboard. Having
to remove SMD resistors is not so easy and also voids the warranty... *NOT
A SMART DESIGN DECISION*. I think the combination of resistors on the I2C
device breakout board 2.2K and the Arduino Due 1.1K has knocked the current
below the 3mA minimum for the High side. Because the values are low they
have to be removed or bypassed.

As a side note: I read on a Raspberry PI forum that there is some sort of
voodoo they do with software that somehow changes resistance values for
pull-ups on board or maybe some other technique. As far as I know there is
no way to control resistance values via software unless one uses what is
called a "digipot (digital potentiometer)." As far as I can tell, there is
nothing like that on the Raspberry PI 4 board. There are other methods to
terminate serial buses but nothing that I've heard of that is "software
selectable." Like I said, it sounds like voodoo but somehow it worked!
Unfortunately, it would take too much effort to make a raspberry PI
function like a microcontroller. It's doable, but I don't want to waste too
much time attempting to reinvent the wheel if you know what I mean...