arduino/Arduino #9802: errors on RAMPS-FD V2.1A by install Marlin 2.0.x

swatkxs, 2020-02-25:
Compiling with Arduino IDE, shows error 
And I have change <configuration.h> #define TEMP_SENSOR_PROBE 0 => #define TEMP_SENSOR_PROBE 1
I also modified <pin_RAMPS_FD_V1.h>
 added 
#if HOTENDS == 1 && TEMP_SENSOR_PROBE
  #define TEMP_PROBE_PIN   2   // TEMP_1_PIN
#endif

> In file included from C:\Users\Automan\AppData\Local\Temp\arduino_build_82994\sketch\src\feature\probe_temp_compensation.cpp:27:0:

C:\Users\Automan\AppData\Local\Temp\arduino_build_82994\sketch\src\feature\probe_temp_compensation.h:60:94: error: could not convert '0.0f' from 'float' to 'XYZval<float>::<anonymous union>'

     static constexpr xyz_pos_t park_point = { PTC_PARK_POS_X, PTC_PARK_POS_Y, PTC_PARK_POS_Z };

                                                                                              ^

C:\Users\Automan\AppData\Local\Temp\arduino_build_82994\sketch\src\feature\probe_temp_compensation.h:60:94: error: too many initializers for 'const xyz_pos_t {aka const XYZval<float>}'

"C:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\libraries\\Wire\\src" "-IC:\\Users\\Automan\\Documents\\Arduino\\libraries\\TMCStepper\\src" "-IC:\\Users\\Automan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\libraries\\SPI\\src" "-IF:\\arduino-1.8.12\\libraries\\LiquidCrystal\\src" "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_82994\\sketch\\src\\feature\\twibus.cpp" -o "C:\\Users\\Automan\\AppData\\Local\\Temp\\arduino_build_82994\\sketch\\src\\feature\\twibus.cpp.o"
使用 1.0  版本的库 Wire 在文件夹： C:\Users\Automan\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\Wire 
使用 0.6.2  版本的库 TMCStepper 在文件夹： C:\Users\Automan\Documents\Arduino\libraries\TMCStepper 
使用 1.0  版本的库 SPI 在文件夹： C:\Users\Automan\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\SPI 
使用 1.0.7  版本的库 LiquidCrystal 在文件夹： F:\arduino-1.8.12\libraries\LiquidCrystal 
exit status 1
为开发板 Arduino Due (Programming Port) 编译时出错。

>

per1234, 2020-02-25:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2020-02-25 by per1234

swatkxs, 2020-02-25:
sorry, I think this is a feature request. Because of the Marlin firmware 2.0.x haven't made the RAMPS-FD V2 a probe temp-sensor yet.

per1234, 2020-02-25:
This is the bug tracker for official Arduino software. Marlin is a 3rd party project that is completely separate from the official Arduino development work. So feature requests for Marlin are off topic here.

ellensp, 2020-08-05:
Issue is in arduino ide, marlin just triggers it. Something broken in DUE framework.