arduino/Arduino #9806: Arduino 1.8.2 Boards Manager has only AVR package to choose from.  Need SAM core.

Europa46, 2020-02-25:
In a fresh installation of Arduino 1.8.12 the boards manager has only one core pack shown for the AVR boards.  I need to install the SAM core to utilize a DUE board but that choice is not available..

facchinm, 2020-02-26:
HI @Europa46 ,
do you get any warning in the console when you open the board manager (like `unable to download index from downloads.arduino.cc/...` ?

Europa46, 2020-03-13:
Sorry,  didn't see this until now.
The problem was obscure,  but it was caused by Java not having a particular
Certification installed.  The security system here would not go to that
site.
However,  now that I have access the SAM package wad downloaded but still
has issues with the installation of the            eabi-none-... tools.
Incorrect path,  I believe.