arduino/Arduino #9807: Change the name of library in Arduino

KorneliusThomas, 2020-02-26:
kornelius_thomas@yahoo.com

Please change the name of my library in Arduino.
Previous name:   PIDcontrollersModular
New name: PID controllers Modular Professional
Library's repository:

https://github.com/KorneliusThomas/PIDcontrollersModularProfessional

per1234, 2020-02-26:
Hi @KorneliusThomas.
I see you created a new repository for the new library name. We can remove the PIDcontrollersModular library from the Arduino Library Manager index and add the new PID controllers Modular Professional library, but first you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file in the new repository.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

KorneliusThomas, 2020-02-26:
I have just created required release in my repository.

regards
Kornelius 

    W środa,

 
 

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2020-03-02:
Done.

closed 2020-03-02 by cmaglie