arduino/Arduino #9813: SparkFun Clock Generator Arduino Library

edspark, 2020-02-27:
Hello,

We have a [new library ](https://github.com/sparkfun/SparkFun_Clock_5P49V60_Arduino_Library) to be added to the Arduino Library Manager, please add at your earliest convenience. 

Thank you!

per1234, 2020-02-27:
Hi @edspark. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

edspark, 2020-02-27:
So sorry that, I knew I was forgetting something! Release - released!

cmaglie, 2020-03-02:
Done.

closed 2020-03-02 by cmaglie