arduino/Arduino #9816: even the blink or bareminimum sketh not compiling in arduino IDE

pro-prodipto, 2020-02-29:
here i am pasting the error messages:

Arduino: 1.8.12 (Windows 10), Board: "Arduino Uno"

avr-gcc: error: @C:\WINDOWS\TEMP\cc8W9IGU: Invalid argument

avr-gcc: fatal error: no input files

compilation terminated.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2020-03-10:
Hi @pro-prodipto ,
it looks like you don't have permissions to write (or read) in a folder needed by the compiler itself.
May you try:
* compiling a sketch with antivirus deactivated?
* compiling a sketch after launching the IDE as Administrator?

Thanks

per1234, 2020-08-04:
Hi @pro-prodipto. Did you ever get the chance to try the things recommended by facchinm? If so, what were the results?

TsuyoEcchi, 2021-07-14:
> Hi @pro-prodipto ,
> it looks like you don't have permissions to write (or read) in a folder needed by the compiler itself.
> May you try:
> 
> * compiling a sketch with antivirus deactivated?
> * compiling a sketch after launching the IDE as Administrator?
> 
> Thanks

Works for me, very Thanks!!

pro-prodipto, 2021-09-30:
> facchinm

closed 2021-09-30 by pro-prodipto

qgz456zhong, 2023-02-15:
> 

十分谢谢你，我正是这个问题，十分感谢你！！