arduino/Arduino #9817: OSX on Mac Pro IDE version 1.8.12 crash

anoracknophobia, 2020-02-29:
crashed while dragging from one screen to a second screen on a dual screen system.

… also had crashed earlier after being left unattended (after opening a file) for approximately
30 minutes i.e.  no other editing of the file had occurred.

SnapeFung, 2020-03-01:
I have the same issue, and my Mac version is Catalina 10.15.3 Arduino version is 1.8.12 too.

per1234, 2020-03-02:
>crashed while dragging from one screen to a second screen on a dual screen system

That sounds like https://github.com/arduino/Arduino/issues/9764

>crashed earlier after being left unattended (after opening a file) for approximately 30 minutes

That sounds like https://github.com/arduino/Arduino/issues/9747

schreinerman, 2020-03-03:
same at my Mac with Catalina... Arduino IDE crashes randomly, so it is impossible to work with it

golfredom, 2020-03-03:
It crashes, Chen trying to dragón an IDE window or serial monitor window to the extended monitor.

AlanTwigg, 2020-03-04:
I have the same issue IDE 1.8.12. Mac OS 10.15.3, dual monitor system

loganduffield, 2020-03-06:
same issue. Mac OS 10.15.3,  IDE 1.8.12. Happens when dragging to extended monitor. Reverted to IDE 1.8.10 and no issue

StefanoSGI, 2020-03-11:
Same issue, works well if I disconnect the 2nd display but that's not practical.
I'm on macOS Catalina, I can reproduce the issue on 1.8.11, 1.8.12 and hourly (`9 March 2020 3:34:57 GMT`).

jeffreyspaceboy, 2020-03-24:
Same issue. Mac OS 10.15.4. IDE 1.8.12. Any time the window is moved or resized.

Vandewaetere, 2020-03-30:
Confirmed here as well. IDE 1.8.10 does not have the issue.
Can be downloaded here: https://www.arduino.cc/download_handler.php?f=/arduino-1.8.10-macosx.zip

facchinm, 2020-03-30:
We are aware of all the issues related with OpenJDK being bundled with 1.8.12 on OSX, but there are only two ways to tackle this (unless Apple reverts its notarization policy):
- fixing AdoptJDK builds (not really our expertise)
- switching to Java14 (with is notarized by Oracle, and *should* be [ok to redistribute](https://www.oracle.com/a/tech/docs/jdk14-lium-2020-03-17.pdf) )

Said that, 1.8.10 will stop working as soon as Apple enforces notarization so it's not a valid long term solution. Any idea is appreciated :slightly_smiling_face:

anoracknophobia, 2020-03-30:
Well thanks for rhe update.  I didn’t want to but I guess I’ll have to give embedXcode along with xcode a try.

regards 
John

StefanoSGI, 2020-03-30:
@facchinm Just an idea... maintain a single code base: go all in with https://www.arduino.cc/pro/arduino-pro-ide and add a "hobby" persona for the original IDE audience.

RyuuzakiJulio, 2020-03-31:
unless Apple reverts its notarization policy?
So Arduino for Mac is dead? Crashes all over the place. Why can't you guys do just a normal app that we can download on the App Store? Or at least not with so many funky frameworks that are not compatible with Mac.

cmaglie, 2020-04-03:
Wait wait wait:

> Said that, 1.8.10 will stop working as soon as Apple enforces notarization so it's not a valid long term solution.

This is not true. Apple has already enforced strict notarization policy, but this is for applications not yet released. Since IDE 1.8.10 has been already notarized it will continue to work also in the future.

So, for now, using IDE 1.8.10 it's the best workaround available.

mctong8664, 2020-04-26:
Hi,

Found a workaround. Go to System Preference -> Display -> Arrangement. Move the white horizontal bar to secondary display. This basically swap the secondary display become primary when 2nd monitor is connected.

Look like Arduino 1.8.12 can only work in primary screen. I just want my Arduino display to work properly on larger monitor. I don't move Arduino display to another screen. So this workaround is fine with me.

Hope this help.

anoracknophobia, 2020-04-26:
A great find if this works, well done.

obdevel, 2020-04-26:
It's getting worse for me. macOS 10.13.6, Arduino IDE 1.8.12. I'm not using a secondary display.

The latest High Sierra update breaks the 1.18.12 IDE in a similar fashion to Catalina, e.g.
- frequent multi-second hangs in the editor
- often results in a corrupted or 'torn' editor window
- sometimes results in a complete hang of the Mac desktop, requiring a hard power-off 

This is in addition to the ongoing issues:
- Library Manager and Boards Manager takes several minutes to process an updated library or boards
- 10-20 second hangs on Tools and File menu

Given the popularity and size of the Arduino user base, not mention the bad publicity, why aren't Apple and Oracle taking more of an interest in this ?

1.8.10 seems more stable.

dirkx-reno-2020, 2020-04-30:
Another work around is selecting the option 'Displays have separate Spaces' in System Preferences -> Mission Control.

obdevel, 2020-05-02:
Having hard rebooted my Mac for the tenth time today, I feel like a rant. Even 1.8.10 causes screen corruption frequent and hard lockups. I can sometimes start a terminal and type the reboot command blind. fwiw my Mac is otherwise typically bulletproof. Only the Arduino IDE causes this behaviour.

I'm more than willing to help with any debugging. Just provide details and I will test and feedback.

High Sierra macOS 10.13.6.

BillyGriffiths, 2020-05-18:
> Hi,
>
> Found a workaround. Go to System Preference -> Display -> Arrangement. Move the white horizontal bar to secondary display. This basically swap the secondary display become primary when 2nd [...]

Good find, thanks! This was driving me nuts!!

BillyGriffiths, 2020-05-18:
> Another work around is selecting the option 'Displays have separate Spaces' in System Preferences -> Mission Control.

This did not work for me on `Catalina 10.15.4`. IDE crashes the moment I draw any window to secondary.

closed 2020-06-12 by facchinm

anoracknophobia, 2020-06-17:
> Closed #9817 <https://github.com/arduino/Arduino/issues/9817> via #10357 <https://github.com/arduino/Arduino/pull/10357>.

Today (17th June 2020), and for the last few days  IDE 1.8.12  still randomly crashing (while on main display)   OSX 10.15.5

cmaglie, 2020-06-17:
We released IDE 1.8.13 with the fix -> https://www.arduino.cc/en/Main/Software

PR #10357 (2020-06-12): Updated appbundler to 1.0ea-arduino8 (fix crash on Macosx Catalina + multi monitor setup)
As discussed in #10234 building Appbundler against MacOSX SDK 10.9 solved the crash on macosx catalina + multi-monitor setup.

Supersedes #10234
Fix #9926 
Fix #9817 
Fix #10302 
Fix #10283 
Fix #10104 
Fix #9957 
Fix #9828 
Fix #9460