arduino/Arduino #9818: class String' has no member named 'remove'

Jaccccc, 2020-02-29:
I am trying to make a version of Michael reeves' laser turret and keep getting the error message that class String' has no member named 'remove'. I have tried the other tutorials to solve this, but this ismy first arduino project and don't know what most people are talking about! Any help would be great! 
This is the code i am trying to upload.

#include <Servo.h>

Servo serX;
Servo serY;

String serialData;

void setup() {

  serX.attach(11);
  serY.attach(10);
  Serial.begin(9600);
  Serial.setTimeout(10);
}

void loop() {
  //lol
}

void serialEvent() {
  serialData = Serial.readString();
  serX.write(parseDataX(serialData));
  serY.write(parseDataY(serialData));
}

int parseDataX(String data){
  data.remove(data.indexOf("Y"));
  data.remove(data.indexOf("X"), 1);
  return data.toInt();
}

int parseDataY(String data){
  data.remove(0,data.indexOf("Y") + 1);
  return data.toInt();
}

deepaklorkhatri007, 2020-02-29:
Hi @Jaccccc, I tried your code and it's compiling just fine for me. Can you please turn on the verbose for compilation and send me the log of the compilation, According to [this post](https://forum.arduino.cc/index.php?topic=294419.0) your IDE might be using a different path for `WString.h` file but I need the log to get a better idea for your situation.

For turning on the verbose please follow these steps:
1. Go to File -> Preference.
2. Turn on the options for verbose like shown in the screenshot and then press OK.
[image]

Jaccccc, 2020-02-29:
I think this is the right thing. Im on raspbian, if that helps too,

/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/cores/arduino -I/usr/share/arduino/hardware/arduino/variants/standard -I/usr/share/arduino/libraries/Servo /tmp/build17535306437653390804.tmp/sketch_feb28f.cpp -o /tmp/build17535306437653390804.tmp/sketch_feb28f.cpp.o

matthijskooijman, 2020-02-29:
I suspect you might be using a (third party) core with a broken or maybe older version of the String class. It would help to know what board you are compiling for.

However, this issue tracker is intended for bugs and feature requests, not for  getting help with your own projects (better use the forum for that). Of course, it might not always be easy to know whether there is a bug in the IDE or in your own code...

Oh, just read your last post - you're using the Raspbian-supplied version of the IDE, which is *ancient* (1.0.5). You really should be installing the latest IDE downloaded from arduino.cc.

So you're indeed using an older version of the String class. Upgrading the IDE should fix this, so I'm closing this issue.

closed 2020-02-29 by matthijskooijman