arduino/Arduino #9823: [Language Reference] Missing EEPROMRead code

Blaster1920, 2020-03-01:
Hi all,
just noticed that the code present [here](https://www.arduino.cc/en/Tutorial/EEPROMRead) is not present anymore.
Could you please update the page?
Thanks.

Similar issue: #7595

thekunalsaini, 2020-03-02:
Is this bug resolved? @per1234

Blaster1920, 2020-03-02:
Looks like it's not yet.

thekunalsaini, 2020-03-03:
Then how to make it solved any suggestion?

per1234, 2020-03-03:
@thekunalsaini the change needs to be made in a place that is not available for contributions from the community. Only an Arduino employee with the correct access privileges can fix this.

ChristianYumul2, 2020-03-11:
I use NotePad/Notepad+ for my Mac/Linux/Mac development work. It’s a shame that people who don’t use it get this awesome feature – it helps me easily get stuff in sync because the Mac App Store makes it easy to find the good stuff. The Mac app that allows you to use the full text of <a href="https://brighterguide.com">brighterguide notepad</a> is pretty awesome and the fact that it doesn’t just open your text editor automatically (so that you don’t have to wait 10 months to get to NotePad and then get it installed, to begin with) saves a ton of my time by having to manually click on a note if I need to. On a related note for those who don’t want their notepad++ app to work, you do need to create multiple notepad++ notes on the first of the week and then have it auto pad on the second day by the end of the week.

frafra, 2020-03-13:
Here is the code:
```c
/*
 * EEPROM Read
 *
 * Reads the value of each byte of the EEPROM and prints it
 * to the computer.
 * This example code is in the public domain.
 */

#include <EEPROM.h>

// start reading from the first byte (address 0) of the EEPROM
int address = 0;
byte value;

void setup() {
  // initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
}

void loop() {
  // read a byte from the current address of the EEPROM
  value = EEPROM.read(address);

  Serial.print(address);
  Serial.print("\t");
  Serial.print(value, DEC);
  Serial.println();

  /***
    Advance to the next address, when at the end restart at the beginning.

    Larger AVR processors have larger EEPROM sizes, E.g:
    - Arduno Duemilanove: 512b EEPROM storage.
    - Arduino Uno:        1kb EEPROM storage.
    - Arduino Mega:       4kb EEPROM storage.

    Rather than hard-coding the length, you should use the pre-provided length function.
    This will make your code portable to all AVR processors.
  ***/
  address = address + 1;
  if (address == EEPROM.length()) {
    address = 0;
  }

  /***
    As the EEPROM sizes are powers of two, wrapping (preventing overflow) of an
    EEPROM address is also doable by a bitwise and of the length - 1.

    ++address &= EEPROM.length() - 1;
  ***/

  delay(500);
}
```

Source: https://web.archive.org/web/20190714024854/https://www.arduino.cc/en/Tutorial/EEPROMRead

sterretjeToo, 2020-05-30:
Why is this closed? Providing the source code anywhere except where it's expected is not solving the issue.

The page still does not show the source code.

per1234, 2020-05-30:
@sterretjeToo this issue is still open. It will only be closed once the problem has been fixed.

sterretjeToo, 2020-05-30:
Ah, you closed another one. Sorry that I missed that.

closed 2020-09-29 by dcuartielles

frafra, 2020-09-29:
@dcuartielles has it been fixed?

per1234, 2020-09-29:
@frafra the issue has been fixed in the source content the tutorials are generated from. Because the PR for the fix was linked to this issue report (thanks Blaster1920!), GitHub automatically closed this issue when that PR was merged.

However, the fix hasn't yet been published to the website.

After it's published, we'll be sure to carefully check to make sure the issue truly has been resolved.

per1234, 2020-10-01:
The fix has now been published to the EEPROM Read tutorial:
https://www.arduino.cc/en/Tutorial/LibraryExamples/EEPROMRead#code

sterretjeToo, 2020-10-16:
Confirmed; indeed solved.