arduino/Arduino #9828: Arduino IDE 1.8.12 crashes upon startup Mac OS X 10.15.3

usaszerelem, 2020-03-03:
Clean uninstalled and reinstalled. I load my sketch, try to do anything with it and crashes. Here is the crash dump.

[ArduinoIDECrash.txt](https://github.com/arduino/Arduino/files/4279409/ArduinoIDECrash.txt)

AlanTwigg, 2020-03-04:
#metoo, IDE 1.8.12 & Mac OS 10.15.3

Arduino quit unexpectedly when opening the Serial Monitor.
**Crashed Thread:        19  Java: Java2D Queue Flusher**

Process:               Arduino [16757]
Path:                  /Users/USER/Desktop/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.8.12 (1.8.12)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [16757]
User ID:               501

Date/Time:             2020-03-04 18:08:30.876 +0000
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Bridge OS Version:     4.2 (17P3050)
Anonymous UUID:        B0D2750B-5588-DF52-C8C1-92D94D355F63

Time Awake Since Boot: 340000 seconds

System Integrity Protection: enabled

Crashed Thread:        19  Java: Java2D Queue Flusher

Exception Type:        EXC_BAD_INSTRUCTION (SIGABRT)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff7108725a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff710875d0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff398a2d0b __CFRunLoopServiceMachPort + 322
3   com.apple.CoreFoundation      	0x00007fff398a18e7 __CFRunLoopRun + 1695
4   com.apple.CoreFoundation      	0x00007fff398a0bd3 CFRunLoopRunSpecific + 499
5   com.apple.HIToolbox           	0x00007fff383f665d RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff383f639d ReceiveNextEventCommon + 600
7   com.apple.HIToolbox           	0x00007fff383f6127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff36a66ba4 _DPSNextEvent + 990
9   com.apple.AppKit              	0x00007fff36a65380 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
10  libosxapp.dylib               	0x0000000107204d2e -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 121
11  com.apple.AppKit              	0x00007fff36a5709e -[NSApplication run] + 658
12  libosxapp.dylib               	0x0000000107204b73 +[NSApplicationAWT runAWTLoopWithApp:] + 219
13  libsplashscreen.dylib         	0x00000001071d1b16 __SplashInitPlatform_block_invoke + 70
14  JavaNativeFoundation          	0x00007fff3e087ff9 +[JNFRunLoop _performCopiedBlock:] + 15
15  com.apple.Foundation          	0x00007fff3bf7f8ab __NSThreadPerformPerform + 254
16  com.apple.CoreFoundation      	0x00007fff398a2b21 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
17  com.apple.CoreFoundation      	0x00007fff398a2ac0 __CFRunLoopDoSource0 + 103
18  com.apple.CoreFoundation      	0x00007fff398a28d4 __CFRunLoopDoSources0 + 209
19  com.apple.CoreFoundation      	0x00007fff398a1740 __CFRunLoopRun + 1272
20  com.apple.CoreFoundation      	0x00007fff398a0bd3 CFRunLoopRunSpecific + 499
21  libjli.dylib                  	0x0000000100255621 CreateExecutionEnvironment + 875
22  libjli.dylib                  	0x0000000100251c1b JLI_Launch + 1878
23  cc.arduino.Arduino            	0x000000010022f166 launch + 8838
24  cc.arduino.Arduino            	0x000000010022cd9f main + 143
25  cc.arduino.Arduino            	0x000000010022cd04 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff710889be __ulock_wait + 10
1   libsystem_pthread.dylib       	0x00007fff7114c315 _pthread_join + 347
2   libjli.dylib                  	0x0000000100255d61 ContinueInNewThread0 + 129
3   libjli.dylib                  	0x0000000100254453 ContinueInNewThread + 170
4   libjli.dylib                  	0x0000000100252881 JLI_Launch + 5052
5   cc.arduino.Arduino            	0x000000010022f166 launch + 8838
6   cc.arduino.Arduino            	0x000000010022cd9f main + 143
7   libjli.dylib                  	0x00000001002563cc apple_main + 84
[... 1019 lines cut ...]
shared memory                      680K       18 
===========                     =======  ======= 
TOTAL                              4.0G     2255 
TOTAL, minus reserved VM space     3.8G     2255 

Model: Macmini8,1, BootROM 1037.80.53.0.0 (iBridge: 17.16.13050.0.0,0), 6 processors, 6-Core Intel Core i5, 3 GHz, 16 GB, SMC 
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E3
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2667 MHz, Micron, 8ATF1G64HZ-2G6E3
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Dec  6 2019 11:17:27 version 9.112.4.0.32.5.41 FWID 01-ac611e8a
Bluetooth: Version 7.0.3f5, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Network Service: iPad, Ethernet, en9
PCI Card: pci11c1,5901, sppci_ieee1394openhci, Thunderbolt@197,0,0
USB Device: USB 3.1 Bus
USB Device: CP2102 USB to UART Bridge Controller
USB Device: Cable Matters USB-C Video Cable
USB Device: iPad
USB Device: Apple T2 Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac mini, Apple Inc., 47.1
Thunderbolt Bus: Mac mini, Apple Inc., 47.1
Thunderbolt Device: Thunderbolt to FireWire Adapter, Apple Inc., 1, 5.6

solick, 2020-03-06:
> Clean uninstalled and reinstalled. I load my sketch, try to do anything with it and crashes. Here is the crash dump.
> 
> [ArduinoIDECrash.txt](https://github.com/arduino/Arduino/files/4279409/ArduinoIDECrash.txt)

same for me. Worked until I updated to Catalina

sskrishnan, 2020-03-08:
Facing same issue. Worked with previous OS.

per1234, 2020-03-08:
The temporary solution is to use Arduino IDE 1.8.10, the last version before OpenJDK was introduced. You can download it here:
https://www.arduino.cc/en/Main/OldSoftwareReleases

I'm not very knowledgeable on this subject, but I believe you will encounter the
>"Arduino" can't be opened because Apple cannot check it for malicious software.

issue with Arduino IDE 1.8.10 due to it not being notarized. There is an explanation of the workaround for that here: https://github.com/arduino/Arduino/issues/9314#issuecomment-541461214

CluEleSsUK, 2020-03-08:
I am having a similar bug on Catalina - the splash screen opens and it runs initialisation etc, but once the splash screen closes, the IDE never appears.

I have tried 1.8.10 as per @per1234 's suggestion, but this also did not work for me.
I can't find the logs that OP links, and in fact my /Users/me/Library/Arduino15/logs/application.log is completely empty - should I be looking elsewhere?

screenmac, 2020-03-09:
I have new 2019 iMac. I update new Java and download latest Arduino.
No blinking Arduino Uno LED blinking lights
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

per1234, 2020-03-09:
@screenmac that is off topic for this issue and likely for this issue tracker in general.

This issue tracker is only to be used to report bugs or feature requests. Your topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

CluEleSsUK, 2020-03-09:
> I am having a similar bug on Catalina - the splash screen opens and it runs initialisation etc, but once the splash screen closes, the IDE never appears.
>
> I have tried 1.8.10 as per @per1234 's [...]

I was able to fix this by downloading the latest hourly build, and when I ran into the security bug on opening it (detailed (here)[https://github.com/arduino/Arduino/issues/9713]), I just right click -> open on the icon (to bypass the Apple check)

yepher, 2020-03-16:
Downgrading to [Arduino IDE 1.8.10](https://www.arduino.cc/en/Main/OldSoftwareReleases) got me back up and running.  I am on OS X `10.15.3 (19D76)`. I've not tried the latest hourly build

facchinm, 2020-03-16:
@gvarisco it looks like Apple is not enforcing the notarization yet... Do you have any insights? Getting rid of the bugs provoked by AdoptJDK would be quite important.

konvader, 2020-03-17:
Had the same issue. Going back to 1.8.10 fixed it for the moment.

yepher, 2020-03-17:
So active editor is null in that case?
[Editor ed = base.getActiveEditor();](https://github.com/arduino/Arduino/blob/master/app/src/cc/arduino/contributions/ContributionsSelfCheck.java#L128)

theDontKnowGuy, 2020-03-21:
Here is why it happens and how to work around it until an official solution is introduced:
When you place the IDE on one screen and the Serial monitor on another, any attempt to move between the two windows or dragging the serial monitor crashes the application immediately.
Two avoid this, place both windows on the same screen, and you shouldn't encounter these crashes.

jenschr, 2020-03-27:
This bug also happens when attaching to a new screen or changing the video output. As an educator that is constantly switching between screens while teaching Arduino, this is a very annoying problem. Arduino IDE will crash every time I toggle between mirroring my own screen and using at as an external screen.

andims, 2020-04-04:
same here.. 
IDE crashes when having more than one screen, works for a while in the second monitor but when I try to open another kind of window (serial monitor i.e) its crashes suddenly

simple2hack, 2020-04-20:
Downgrading to Arduino IDE 1.8.10 also got me back up and running. I am on OS X 10.15.4.  IDE 1.8.12 was unusable once I tried to load sketches.  OK for checking libraries.  Just crashed when I tried to load a sketch.  I do have an HDMI monitor in addition to my laptop screen.

Chrissy4605, 2020-04-21:
MacOS 10.15.4 same as above. I will try IDE 1.8.10!!! thank you!!!

Brewj, 2020-05-14:
> Here is why it happens and how to work around it until an official solution is introduced:
> When you place the IDE on one screen and the Serial monitor on another, any attempt to move between the two [...]

Exactly I have found if I attempt to move any of the sketch or serial monitor windows to a different screen it crashes.
Mac OS10.15.4 IDE 1.8.12 - Under Mission Control I run that Displays have separate spaces. I'm yet to try disabling that and seeing if it makes a difference

achtenhagen, 2020-05-17:
FWIW, I am running macOS 10.15.4 and it crashes only when using an external display, i.e. moving any Arduino IDE window between displays. Without using the external display, the application has not yet crashed yet for me.

lvangent, 2020-05-17:
Hi,
This is a known problem and has to do with the version of SQL. 
solution: reverting back to IDE 1.8.10. 

Leen van Gent

dansbrazil, 2020-06-10:
MBP 16, macOS Catalina 10.15.4
After installing a board in Arduino, the app crashed and would crash every time I open it up. Tried to reinstall, nothing.
Even tried the 1.8.10 version, nothing.
Then I used App Cleaner to remove the Arduino App, and reinstalled. Voilá! It works!

dansbrazil, 2020-06-10:
After trying to install again the board, closed the app. Trying to open the App again, it keeps crashing.
Used App Cleaner to clean the /Library/Arduino15 folder. Now the App opens. After closing (library is back in place) , the App doesn't open anymore.
I don't really want to keep cleaning the Library folder every time I need to open the Arduino App.

spongefile, 2020-06-10:
I'm having the same problem, MacOS 10.15.5, latest non-beta Arduino version (from last week. Can't find the version number), brand new MacBook Air with nothing else on it yet. At first it worked fine (I added a library for Move38 Blinks and played around with that for a while) and now a few days later (perhaps because I closed the app) the app opens, shows the loading screen, and immediately closes/crashes.

spongefile, 2020-06-10:
Can verify that using AppCleaner to remove the /Library/Arduino15 folder makes it work again. Annoying though.

QimatLuo, 2020-06-10:
Same issue.
Mac 10.15.5, was work for few days. Today I try to make my own library in the Arduino library folder and still works fine. After I restart my computer then Arduino not able to open any more.
Even though remove the Arduino folder or download Arduino again.

QimatLuo, 2020-06-10:
By the way, I don't have any Arduino related folder under /Library.

QimatLuo, 2020-06-10:
Oh, just realize it's ~/Library, not /Library.
After delete the folder then it's able to open now.

QimatLuo, 2020-06-10:
But I cannot open Board Manager anymore.
Cannot download esp8266, still blocked.

QimatLuo, 2020-06-10:
<details><summary>crash report</summary>
<p>
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellJPanel.update(ContributedPlatformTableCellJPanel.java:222)
	at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellRenderer.getTableCellRendererComponent(ContributedPlatformTableCellRenderer.java:47)
	at javax.swing.JTable.prepareRenderer(JTable.java:5729)
	at javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2114)
	at javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2016)
	at javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1812)
	at javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
	at javax.swing.JComponent.paintComponent(JComponent.java:780)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JViewport.paint(JViewport.java:728)
	at javax.swing.JComponent.paintChildren(JComponent.java:889)
	at javax.swing.JComponent.paint(JComponent.java:1065)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
	at java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
	at java.awt.Dialog.show(Dialog.java:1084)
	at java.awt.Component.show(Component.java:1671)
	at java.awt.Component.setVisible(Component.java:1623)
	at java.awt.Window.setVisible(Window.java:1014)
	at java.awt.Dialog.setVisible(Dialog.java:1005)
	at processing.app.Base.openBoardsManager(Base.java:1414)
	at processing.app.Base$10.actionPerformed(Base.java:1441)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
[... 33 lines cut ...]
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:357)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:345)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:763)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
</p>
</details>

facchinm, 2020-06-10:
We had a problem on our package_index.json that caused the error and prevented the IDE to start.
Now the package_index has been fixed, but to recover a working installation you must delete once again the package_index.json with the cache folder and restart the IDE as usual.

Windows:
* remove file C:\Users\YourUsername\AppData\Local\Arduino15\package_index.json
* remove folder C:\Users\YourUsername\AppData\Local\Arduino15\cache

Mac:
* remove file /Users/YourUsername/Library/Arduino15/package_index.json
* remove folder /Users/YourUsername/Library/Arduino15/cache

Linux:
* remove file /home/YourUsername/.arduino15/package_index.json
* remove folder /home/YourUsername/.arduino15/cache

Please note that the problem that prevents the IDE to start has already been fixed and it's ready for the next release, that's the reason why the Nightly/Beta Builds are not affected, and also the reason why we did not detect this problem earlier, sorry about that!

closed 2020-06-10 by facchinm

tony1tf, 2020-06-10:
Thank you @facchinm. A couple of nights ago I was working happily with Arduino 1.8.12 on my MacBook with MacOS 10.15.3, then yesterday I tried to use it and it just loaded the splash screen and crashed as others have said. I deleted the app and reloaded - same problem until I saw your email at the bottom of the page - deleted the files and it is working again.
Tony

dsantiago, 2021-04-19:
Crashes a lot on MAC, why this issue is closed?

PR #10357 (2020-06-12): Updated appbundler to 1.0ea-arduino8 (fix crash on Macosx Catalina + multi monitor setup)
As discussed in #10234 building Appbundler against MacOSX SDK 10.9 solved the crash on macosx catalina + multi-monitor setup.

Supersedes #10234
Fix #9926 
Fix #9817 
Fix #10302 
Fix #10283 
Fix #10104 
Fix #9957 
Fix #9828 
Fix #9460