arduino/Arduino #9829: Libraries not loading in

makemake5050, 2020-03-03:
exit status 1
'Mouse' not found. Does your sketch include the line '#include <Mouse.h>'?

But I included the library mouse.h

per1234, 2020-03-03:
This error is caused by compiling for an Arduino board that is not compatible with the Mouse library. Only boards with native USB capabilities (e.g., [Leonardo](https://store.arduino.cc/arduino-leonardo-with-headers), [Micro](https://store.arduino.cc/arduino-micro), [MKR boards](https://store.arduino.cc/catalogsearch/result/?q=mkr), [Nano 33 IoT](https://store.arduino.cc/arduino-nano-33-iot)) can be used with the Mouse and Keyboard libraries. If you have one of these boards, then you need select the correct board from the Arduino IDE's **Tools > Board** menu. If you don't have one of these boards, then you can't use the Mouse library.

closed 2020-03-03 by per1234

makemake5050, 2020-03-03:
I was using an arduino nano witch is in your list

per1234, 2020-03-03:
No, the [Nano 33 IoT](https://store.arduino.cc/arduino-nano-33-iot) is on my list. The[ original Arduino Nano](https://store.arduino.cc/arduino-nano) is not on the list because it doesn't have native USB capabilities.

makemake5050, 2020-03-03:
Is there any way to get it to work

per1234, 2020-03-03:
No. The only way is to buy a different board.