arduino/Arduino #9831: Arduino Nano 33 BLE IDE is not handling port changes on upload correctly

derekchall, 2020-03-03:
The boot loader on the nano 33 BLE changes the port used when uploading but the Arduino IDE is not coping with the change correctly. On my system the Arduino uses port 8, with port 8 selected in the tools menu when an upload is performed the port changes to port 6 and the upload continues as normal, at the end of a successful upload the port changes back to 8 but does not get selected in the tools menu(this means that if you don’t manually select the port again the next upload will fail) , surly  port 8 should automatically be selected in the tools menu as then no-one would even know or care about the changing com ports.
I am using the installed version of Arduino IDE 1.8.12 on windows 10
I was directed to post here from the forum.

Thanks.

pavitter-bluestone, 2020-06-12:
I'm experiencing a similar issue. My Arduino is connected to COM Port 6 when i press upload, during upload it changes to Port 7 and then i need to manually reselect Port 6. Otherwise i cannot open the Serial Monitor.

bavanyo, 2021-08-13:
I'm having the same issue.
Has anyone found a solution yet?