arduino/Arduino #9832: [Library Manager] Please add

peterus, 2020-03-03:
Please add my lib to the library manager.

https://github.com/peterus/APRS-IS-Lib

Thanks!
Peter

per1234, 2020-03-03:
Hi @peterus. You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

peterus, 2020-03-03:
Hi @per1234 
Sorry, totally missed to create a tag. Was creating it now ;)

peterus, 2020-03-10:
please add https://github.com/peterus/APRS-Decoder-Lib too :)
thank you!

cmaglie, 2020-03-20:
Done.

closed 2020-03-20 by cmaglie