arduino/Arduino #9837: gcc11 loses avr-backend - without renovation

mikrocoder, 2020-03-05:
Hello Arduino friends and all other avr-gcc users

I don't know if the necessary avr-backend renovation has got around yet?

[https://gcc.gnu.org/PR92729](url)
[https://gcc.gnu.org/ml/gcc-patches/2019-09/msg01256.html](url)
[https://www.bountysource.com/issues/84630749-avr-convert-the-backend-to-mode_cc-so-it-can-be-kept-in-future-releases](url)

The current situation is the following. If the avr-backend is not renovated, the avr support from gcc11 is out.

There is currently a fundraiser on bountysource which could pay a programmer to do this. The current target amount is 5000,- Dollar. Whether it will be the same programmer who made the necessary changes for Motorola is not certain. But it is possible.

I would also like to ask if the Arduino team has already thought about this? After all, avr-gcc including C++ programming is used intensively. Can Arduino.cc imagine supporting the renovation in any way?
Either by hiring a programmer for a short time or by supporting the project financially? Or in some other way?

--- deutsch ------------------------------------------------------

Hallo Arduino Freunde und alle anderen avr-gcc Nutzer.

Ich weiß nicht ob sich die notwendige avr-backend Renovierung schon rumgesprochen hat?

Die aktuelle Situation ist folgende. Wenn das avr-backend nicht renoviert wird fliegt die avr Unterstützung ab gcc11 raus.

Es gibt derzeit eine Spendensammlung auf bountysource womit ein Programmierer bezahlt werden könnte der das macht. Die derzeit angestrebte Summe ist 5000,- Dollar. Ob es der gleiche Programmierer macht der die notwendigen Änderungen schon für Motorola gemacht hat steht nicht fest. Wäre jedoch denkbar.

Ich möchte hiermit auch fragen ob sich das Arduino Team darüber schon gedanken gemacht hat? Schließlich wird der avr-gcc inkl. C++ Programmierung intensiv genutzt. Kann sich Arduino.cc vorstellen die Renovierung in irgendeiner Form zu unterstützen?
Entweder stellt man kurzzeitig einen Programmierer ab oder unterstützt die Sache finanziell? Oder irgendwie anders?

mikrocoder, 2020-03-14:
Hello,

can you expect a comment on the topic from the Arduino.cc team or are they not reading this?

Floessie, 2020-03-27:
Meanwhile in [LLVM 11...](https://llvm.org/docs/ReleaseNotes.html#changes-to-the-avr-target)

per1234, 2020-12-24:
Some good news on the Arduino Developer's mailing list:
https://groups.google.com/a/arduino.cc/g/developers/c/62z114HR9w0/m/LfEOlxGsBAAJ
>Senthil Kumar from Microchip/Atmel themselves is working on
this issue now and the AVR backend will definitely be saved.

So I'll go ahead and close this. Thanks to everyone who supported this initiative!

closed 2020-12-24 by per1234