arduino/Arduino #9842: label:"Component: Documentation" Uno WiFI Rev 2 - Tech Specs - Input voltage limit

Bylund, 2020-03-06:
Hi, according to the Arduino Uno WIFI Rev 2 tech specs the upper input voltage limit is 12V. Compared to other boards this seem low. According to the schematics it is using a MPM3610  step-down converter capable of up to 21V input. I am guessing the actual Input Voltage (limit) should be 6-20V like the other UNO's?

dushyant18033, 2020-03-12:
Hi, I checked out the [datasheet](https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MPM3610GQV/document_id/2090) for MPM3610. But there is a table on page 16 of the datasheet which gives the various component specifications to be followed in the power circuit for a particular limit of Vin, which gives some hint that maybe Uno Wifi Rev2 is using the specifications for Vin=12V probably because MPM3610 works much more efficiently for Vin limit of 12V (checkout the typical characteristics starting at page 5 for more information).
So probably the tech-specs on the website is correct in my opinion. Please verify with the schematic for component values used for Vin=12V.

Bylund, 2020-11-29:
Hello @alranel. Thank you for the update. We wanted to recommend using the Uno WIFI Rev 2 in automotive applications up to 15V, now we can. Note that as of today there are still no updates to the Uno WIFI Rev 2 spec on the Arduino website.

per1234, 2020-11-29:
Hi @Bylund. I hope you won't mind if I reopen this. As you pointed out, the tech specs have not yet been updated as planned, so I think it's best to keep this open so it isn't forgotten.

per1234, 2022-07-17:
Note to the assignee: please make sure any update is made to all occurrences of the information:

- https://docs.arduino.cc/hardware/uno-wifi-rev2#panel-teach-spec
- https://store.arduino.cc/products/arduino-uno-wifi-rev2#tech-specs