arduino/Arduino #9853: Basic GLCD(QY-12864BG) Arduino Library

KwangryeolPark, 2020-03-09:
Hi, I developed a library for QY_12864BG Graphic LCD.
Please add this library to the library manager.

Github URL : [https://github.com/KwangryeolPark/GLCD_QY_12864BG](url)

Thank you so much.

per1234, 2020-03-09:
Hi @KwangryeolPark
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

KwangryeolPark, 2020-03-09:
Thank you for helping me.
I edit the Library file as you said me.
Please check it.
Thank you

per1234, 2020-03-09:
The library.properties file looks good. Now all you need to do is either redo the existing tag/release or update the `version` value in library.properties and then create a new tag/release.

KwangryeolPark, 2020-03-09:
https://github.com/KwangryeolPark/GLCD_QY_12864BG/

cmaglie, 2020-03-20:
Done.

closed 2020-03-20 by cmaglie