arduino/Arduino #9855: Cannot Remove Activation Bar after Activation e-mail confirmed by user

RogerioCassares, 2020-03-09:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Please I have Registred and activated de email accout but Activarion Warning Bar until remais and i cannot close.

[image]

How may I proceed, please?

KevinChittilapilly, 2020-07-27:
Has this issue been rectified? Because I would like to work on it if it hasn't

per1234, 2020-07-27:
@KevinChittilapilly thanks for your interest in helping. The code for the Arduino web site is not hosted in a public repository so there is no way for you to work on it.

However, Arduino has a huge amount of open source code that is hosted in public repositories. This offers many opportunities for you to contribute to the Arduino project by helping out with the code.

You can find the repositories listed under these two GitHub organizations:

- https://github.com/arduino
- https://github.com/arduino-libraries

Regards, Per