arduino/Arduino #9856: LCD Output via SoftwareSerial broken on IDE 1.8.10+ (1.8.9 OK)

LKRynearson, 2020-03-09:
I am not positive where this goes and am posting in arduino/Arduino requesting help in moving this to where it belongs.

The short description behavior we are seeing is that when using the SoftwareSerial library instead of the regular TX pin to write to an LCD, text does not seem to print correctly on and the cursor location seems to move around so that characters end up where they shouldn't be. This is new behavior after years of everything working fine with the same product and code (so only the IDE or possibly the library has changed).

The issue is described in detail in [this Arduino forum post](https://forum.arduino.cc/index.php?topic=667347.0) including all hardware, code, etc.

I have verified this issue still exists in the hourly build as of the time of posting. Therefore, the issue exists in 1.8.10, 1.8.11, 1.8.12 and the hourly build as of posting. Version 1.8.9 works as previously / expected. This is on Windows 10, using Arduino Unos, and [this LCD](https://docs.google.com/viewer?url=https%3A%2F%2Fwww.parallax.com%2Fsites%2Fdefault%2Ffiles%2Fdownloads%2F27979-Parallax-Serial-LCDs-Product-Guide-v3.1.pdf).

LKRynearson, 2020-03-30:
In the interim we have found more information - this appears to be related to the AVR version.

On 1.8.9 with AVR version 1.6.23, there is no problem.
On 1.8.12 with AVR 1.8.2, the problem exists. (also 1.8.1)
On 1.8.12 with AVR 1.6.23, there is no problem.

Is there a better way to tag or move this to bring it to the attention of the right people?

Thanks!

facchinm, 2020-03-30:
Hi @LKRynearson ,
the issue looks similar to https://github.com/arduino/Arduino/issues/9653 (too much optimization by new compilers).
Adding `-fno-jump-tables` could help here too; would you mind testing it in your setup? Thanks

LKRynearson, 2020-03-30:
I would love to - can you or another provide some support as to what that means? Is this something to append to the executable when running the IDE (my first guess) or something with how the compiler runs (my second)? I did a quick Google search and am still not sure how to proceed. Thanks!

facchinm, 2020-03-30:
The best way to proceed is probably:
* activate verbose compilation
* find out where the AVR core resides
* add the flag to `compiler.c.flags` and `compiler.cpp.flags` in `platform.txt`

LKRynearson, 2020-05-05:
I found platform.txt and I believe I added the flags in the correct manner - not 100% sure but it looks reasonable to my inexperienced eye. I also added the same flags in 'extra flags' just to see if that changed anything, but it did not. 

The Arduino IDE opens, runs, and compiles, but **no change in the problematic behavior is observed**. Can you verify my flags were set correctly? The code below is copy/paste from my test version of platform.txt. 

`# Default "compiler.path" is correct, change only if you want to override the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os  {compiler.warning_flags} -fno-jump-tables -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects
compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -flto -MMD
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -fno-jump-tables -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto`

The only changes should be in `compiler.c.flags` and `compiler.cpp.flags` as directed.

Thanks!

LKRynearson, 2020-05-05:
I tested removing the `-flto` flags as recommended in https://github.com/arduino/Arduino/issues/9653 and **this did fix the problematic behavior**. I don't know if that means the flags are set wrong but I felt it would likely be helpful information.