arduino/Arduino #9858: bug: text cursor shown at wrong position when typed two-space-width Traditional Chinese Unicode Charactor in comment

yonoodle, 2020-03-10:
text cursor shown at at wrong position when typed two-space-width Traditional Chinese Unicode Charactor in the comments , the charactor occupied 2 space in width , but the cursor only move 1 space , which result in cursor display at middle of charactor shown , or at wrong position then makes the text editing difficult.

please paste these two charactor to IDE for testing  , "測試"  (means "test" in Trad. Chinese)

[image]

ScarySheep, 2020-03-16:
Can't replicate the bug in 1.8.12. What input method are you using?微軟注音?新酷音?
I've tried both typing by myself and pasting your word.

qwec01, 2021-12-26:
> 

When the font does not contain Chinese characters, this bug can be relicated.

per1234, 2021-12-26:
I am still able to reproduce this in Arduino IDE 1.8.19. It is necessary to use specific fonts, as explained in https://github.com/arduino/Arduino/issues/10933

The good news is that I can not reproduce it in Arduino IDE 2.0.0-rc3