arduino/Arduino #9866: file open dialog does not show symlinks

rfuegen, 2020-03-11:
OS: Linux Mint 19.3
Desktop: Cinnamon
Arduino IDE: 1.8.12 (installed via flatpak)

$HOME/Source is symlink to /media/bigdatadisk/Source/
$HOME/Source/Arduino/ contains all projects

the open dialog does not show ANY symlinks (neither Source, nor Downloads, Documents, .. which are all symlinks).

I used Arduino IDE 1.8.5 on Fedora 28 with Cinnamon before, where the symlinks were shown as expected.

the file open dialog in i.e. LibreOffice shows symlinks and even hidden files (.dotfiles).

maybe this is an integration issue?

facchinm, 2020-03-12:
@rfuegen do you have a chance to run the official IDE? It might be a jre problem triggered by flatpak packaging method.

rfuegen, 2020-03-12:
@facchinm yes, installed official IDE, but still no symlinks. 

java -version
openjdk version "1.8.0_242"
OpenJDK Runtime Environment (build 1.8.0_242-8u242-b08-0ubuntu3~18.04-b08)
OpenJDK 64-Bit Server VM (build 25.242-b08, mixed mode)

tried also with

java -version
openjdk version "11.0.6" 2020-01-14
OpenJDK Runtime Environment (build 11.0.6+10-post-Ubuntu-1ubuntu118.04.1)
OpenJDK 64-Bit Server VM (build 11.0.6+10-post-Ubuntu-1ubuntu118.04.1, mixed mode)

but same effect. 

or does it use the java version contained in the tarball?

how can I be of further help?

rfuegen, 2020-03-12:
sorry, my bad, please ignore my previous post. had an issue with mounting the data partition.

It works with the official IDE. so ot really seems to be a packaging issue.

rfuegen, 2020-03-12:
installed the official IDE from tarball, symlinks show up now.

it really seems to be a flatpak packaging issue.

thanks.

facchinm, 2020-03-13:
Probably it's using the JRE bundled with the operating system, while we provide Oracle JRE. CLosing here, please report upstream if needed, thanks!

closed 2020-03-13 by facchinm