arduino/Arduino #9868: Bug: Arduino 1.8.12/1.8.13 Serial Monitor crashing

Jam5, 2020-03-11:
Hi,

Serial Monitor is crashing when I try to open it.
With Arduino 1.8.1 the Serial Monitor works fine.
 
Process:               Arduino [15107]
Path:                  /private/var/folders/*/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               nightly (1.8.13)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [15107]
User ID:               501

Date/Time:             2020-03-12 00:04:33.049 +0100
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        227DE1D7-2DEB-659E-2732-6A97717E33F0

Sleep/Wake UUID:       C2A9D1A6-5008-4726-ABEB-5C09666DD92B

Time Awake Since Boot: 15000 seconds
Time Since Wake:       13000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        19  Java: Java2D Queue Flusher

Exception Type:        EXC_BAD_INSTRUCTION (SIGABRT)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.

Thread 0:: AppKit Thread
0   ???                           	0x00007fff63e82ce6 0 + 140734869548262
1   libjvm.dylib                  	0x000000010be72aa4 Parker::park(bool, long) + 510
2   libjvm.dylib                  	0x000000010bf6df11 Unsafe_Park + 126
3   ???                           	0x000000011982d02a 0 + 4722970666
4   ???                           	0x0000000119663694 0 + 4721096340
5   ???                           	0x0000000118db27a7 0 + 4711983015
6   libjvm.dylib                  	0x000000010bcd79fe JavaCalls::call_helper(JavaValue*, methodHandle*, JavaCallArguments*, Thread*) + 1710
7   libjvm.dylib                  	0x000000010bd0ed3c jni_invoke_nonstatic(JNIEnv_*, JavaValue*, _jobject*, JNICallType, _jmethodID*, JNI_ArgumentPusher*, Thread*) + 748
8   libjvm.dylib                  	0x000000010bd01669 jni_CallVoidMethodV + 248
9   JavaNativeFoundation          	0x00007fff30e399b0 JNFCallVoidMethod + 175
10  libawt_lwawt.dylib            	0x00000001113ecc36 -[CGLLayer drawInCGLContext:pixelFormat:forLayerTime:displayTime:] + 287
11  com.apple.QuartzCore          	0x00007fff37f1cab9 CAOpenGLLayerDraw(CAOpenGLLayer*, double, CVTimeStamp const*, unsigned int) + 1978
12  com.apple.QuartzCore          	0x00007fff37f1c124 -[CAOpenGLLayer _display] + 580
13  com.apple.QuartzCore          	0x00007fff37e7f9f1 CA::Layer::display_if_needed(CA::Transaction*) + 739
14  com.apple.QuartzCore          	0x00007fff37e5e3da CA::Context::commit_transaction(CA::Transaction*, double) + 334
15  com.apple.QuartzCore          	0x00007fff37e5d002 CA::Transaction::commit() + 638
16  com.apple.QuartzCore          	0x00007fff37e98966 CA::Transaction::observer_callback(__CFRunLoopObserver*, unsigned long, void*) + 66
17  ???                           	0x00007fff2c632c58 0 + 140733938084952
18  ???                           	0x00007fff2c632b89 0 + 140733938084745
19  ???                           	0x00007fff2c631773 0 + 140733938079603
20  com.apple.HIToolbox           	0x00007fff2b17865d RunCurrentEventLoopInMode + 292
21  com.apple.HIToolbox           	0x00007fff2b17839d ReceiveNextEventCommon + 600
22  com.apple.HIToolbox           	0x00007fff2b178127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
23  ???                           	0x00007fff297e9eb4 0 + 140733889552052
24  ???                           	0x00007fff297e8690 0 + 140733889545872
25  libosxapp.dylib               	0x000000010fd46328 -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 124
26  ???                           	0x00007fff297da3ae 0 + 140733889487790
27  libosxapp.dylib               	0x000000010fd45f72 +[NSApplicationAWT runAWTLoopWithApp:] + 156
28  libsplashscreen.dylib         	0x000000010fd225c8 __SplashInitPlatform_block_invoke_1 + 360
29  JavaNativeFoundation          	0x00007fff30e3cff9 +[JNFRunLoop _performCopiedBlock:] + 15
30  ???                           	0x00007fff2ed3488b 0 + 140733978986635
31  ???                           	0x00007fff2c6336a6 0 + 140733938087590
[... 971 lines cut ...]
TOTAL                              3.9G     1823 
TOTAL, minus reserved VM space     3.5G     1823 

Model: MacBookPro10,2, BootROM 281.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,5 GHz, 8 GB, SMC 2.6f59
Graphics: kHW_IntelHD4000Item, Intel HD Graphics 4000, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x10F), Broadcom BCM43xx 1.0 (7.21.190.33 AirPortDriverBrcm4360-1415)
Bluetooth: Version 7.0.2f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en4
PCI Card: Apple 57762-A0, sppci_ethernet, Thunderbolt@195,0,0
Serial ATA Device: TS240GJDM720, 240,06 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: FT232R USB UART
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 3, 5.5

anaicagrouver, 2020-03-15:
Can you share the code you tried to run? And what was your baud rate? 
Maybe try to reduce the baud rate

facchinm, 2020-03-16:
@Jam5 is your operating system macOS ? In this case it's a duplicate of one of these (https://github.com/arduino/Arduino/labels/OpenJDK) all related with the JKD we are forced to bundle with OSX due to the notarization problems.

closed 2020-03-16 by facchinm