arduino/Arduino #9869: #line comments added in string literal defined with const char INDEX_HTML1[] = R"=====(

heifau, 2020-03-12:
When compiling a .ino file, somehow #line comments are added in
string literals defined with R"=====( that contain javascript function definitions.

This leads to errors, when the <script> part of the html page is interpreted by the browser/javascript, where "#" is not a valid comment.

If found them in 
C:\Users\..\AppData\Local\Temp\arduino_build_853222\sketch\...ino.cpp
E.g.:
...
const char INDEX_HTML1[] =
R"=====(
<!doctype html>
...
<script>
...
#line 51 "C:\\Users\\....ino"
function myonrestch(xhttp);
...
function myonrestch(xhttp) {
    // sessionStorage.resz = Number(sessionStorage.resz) + 1;
};

heifau, 2020-03-12:
For each function, two line are added: one with #line and wone with "function fnname();".
I noticed it first with version 1.8.10. The upgrade to 1.8.12 didn't help.

matthijskooijman, 2020-03-12:
This is a bit of a known problem. The IDE insert function prototypes at the top of your sketch (which is a feature, so you can use functions before they are declared), but it sometimes puts them in the wrong place. It normally puts them just above the first function declaration. In this case, it sees the javascript functions and puts stuff just before that, breaking your string.

This is hard to fix (though there has been some work to *properly* parse the source instead of the current ctags/regex-based approach, not sure what the status of that is). Regardless, a workaround you could use to put a function prototype (aka forward declaration) *above* your string constant, so the IDE will insert things there.

e.g., put something like:

    void setup();

above your string constant, which forward-declares the setup function. This should not affect the code in any way, but could fix the insertion of the prototypes.

heifau, 2020-03-13:
Thanks   for the good and fast answer.
Ev. I will try another solution, and put the string in a SPIFFS file. You
can close the case.

matthijskooijman, 2020-03-13:
Thanks for the followup.

> You can close the case.

You could have as well, but let me :-)

closed 2020-03-13 by matthijskooijman