arduino/Arduino #9879: [Library manager] Where's the uninstall button?

rtek1000, 2020-03-16:
Is it only possible to install and update?

How is the removal done?

Should the user go to the folder and delete it, or is there a button in the manager?

per1234, 2020-03-16:
We have an answer in the documentation here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-delete-a-library
> ### How can I delete a library?
>
>Open your sketchbook "libraries" folder with your OS file explorer (Windows: Explorer, Mac: Finder, Linux: Nautilus, kfiles...) and delete the folder containing your library.
>
>#### Rationale
>
>No, we don't have a delete button. Libraries managed by the Library Manager are mixed with those you've manually installed, maybe libraries you've written: making a mistake and deleting the wrong library is too easy. That's why trash bins exist. Since the IDE has no knowledge of your trash bin, we didn't implement a "delete" button.

More discussion on the topic at https://github.com/arduino/Arduino/issues/8269 and https://github.com/arduino/Arduino/issues/2878

Since we already have multiple feature requests for this, I'm going to close as duplicate.

closed 2020-03-16 by per1234

rtek1000, 2020-03-16:
Well, It is very strange to be able to install and not to uninstall by the manager, instead of manager we can call it "Installer and Updater".

I would like to comment that there is a lack of refresh from the manager after updating a library. To resolve this, I have to switch between All and Updatable a few times

Thank you.

per1234, 2020-03-16:
>It is very strange to be able to install and not to uninstall by the manager

This has been noted in the previous issue reports. If anyone wants to provide any new and original input on the subject, please make it on https://github.com/arduino/Arduino/issues/2878 so we can keep all the discussion in one place.

>I would like to comment that there is a lack of refresh from the manager after updating a library. To resolve this, I have to switch between All and Updatable a few times

I can reproduce this. Please submit a dedicated issue.