arduino/Arduino #9886: Linux Installer and Gtk errors

zermingore, 2020-03-18:
After upgrading Arduino from version 1.8.10 to 1.8.12 using the install script, I noticed that I could not use terminator anymore (early crashing with a Gtk error as output)

In xterm:
`$> terminator`

> (terminator:12853): Gtk-WARNING **: 09:02:59.417: Could not load a pixbuf from icon theme.
> This may indicate that pixbuf loaders or the mime database could not be found.
> **
> Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
> Aborted

The involved file /usr/share/icons/Adwaita/16x16/status/image-missing.png existed
After a reboot, my window manager (i3) did not start anymore

These did not work:
`$> update-mime-database ~/.local/share/mime`
`$> sudo update-mime-database /root/.local/share/mime`

The fix:
`sudo apt-get install --reinstall shared-mime-info`

I can reproduce the issue re-running the install.sh script and appended the output file of the following:

`sudo bash -x ./install.sh > log_arduino_install_1.8.12 2>&1`

[log_arduino_install_1.8.12.txt](https://github.com/arduino/Arduino/files/4347774/log_arduino_install_1.8.12.txt)

SvenMb, 2020-07-28:
Same problem here with arduino 1.8.13 and it is not only terminator, other GTK-apps have the same problem.

(and thanks for the quick fix!)

wsanders, 2022-04-21:
Still an issue on Fedora 35 with IDE 1.8.19. This problem is upstream (in Java? GDK?) since I, too, see the same messages scrolling in gimp, which is unrelated to Arduino IDE.