arduino/Arduino #9887: I have found an issue with the LCD display Output

Camo2000, 2020-03-18:
I have found an issue with the LCD display Output, it works fine until you add the word  "Mode" as a string. The display the corrupts and i can not see why unless there is a method to control the LED display with text. however if you use any string with the word in it it still corrupts
"Mode"
"Edit Mode"
"Active Mode"
All fail.

String MyTitles[ARRAYSIZE]={"Mode", "Effect", "Pallete", "Sin8", "Sin16", "Speed", "LED Spacing"};
this fails if Mode is in the list

Hope some of you can replicate this.

per1234, 2020-03-18:
Please post your full sketch.

Camo2000, 2020-03-18:
#define ARRAYSIZE 7
String newtit[ARRAYSIZE]={"Apple","Effect","Pallete","Sin8","Sin16","Speed","LED Spacing"};
int iValues[ARRAYSIZE] = {1,2,3,4,5,6,7};
#include <LiquidCrystal.h>
const int rs = 12, en = 11, d4 = 5, d5 = 4, d6 = 3, d7 = 7;// initialize the library by associating any needed LCD interface pin
LiquidCrystal lcd(rs, en, d4, d5, d6, d7);// with the arduino pin number it is connected to

void setup() { 
  Serial.begin(9600); 
  lcd.begin(16, 2); // set up the LCD's number of columns and rows:
  lcd.print("Display Active"); // Print a message to the LCD.
  delay(2000);
}
void loop() {
  int iSelection=0;
  lcd.setCursor(0, 1);
  String message =(String)"A" + iValues[iSelection] + " " + newtit[iSelection];
  Serial.println(message);
  lcd.print(String(message));
  delay(5000);
}

Camo2000, 2020-03-18:
just change the word "APPLE" to "Mode" and the LCD will fail or any word
use int iSelection=0; to select the array space Apple is 0 but you can put the word Mode anywhere in the array and when accessed it fails

per1234, 2020-03-18:
Which board are you using?

Camo2000, 2020-03-18:
it is the standard LCD board 1602A

per1234, 2020-03-19:
Which Arduino board. For example: Uno, Leonardo, Mega, MKRZero, Nano 33 IoT...

Camo2000, 2020-03-19:
Uno R3, but the board seems to be working the LCD only fails if a word containing this sequence "MOD" are sent to the LCD

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/LiquidCrystal/issues/44

closed 2021-03-28 by per1234