arduino/Arduino #9891: [Library Manager] Add PCA9633 to the Library Manager

HendrikVE, 2020-03-19:
Hello,

I know there's already another library for this controller, but I am not happy with the feature set, so I wrote my own library with a more feature complete API.
You can find it here:
https://github.com/HendrikVE/ArduinoPCA9633

BTW: Is "Arduino" in the name of the repository ok? I know from https://forum.arduino.cc/index.php?topic=604471.0 that the Arduino prefix is reserved now. The name declared in my `library.properties` does not contain this prefix.

Thanks

per1234, 2020-03-19:
Hi @HendrikVE. The "Arduino" prefix on the repository name does not pose any problems for addition of your library to the Arduino Library Manager index.

Unfortunately, there is already a library in the index named "[PCA9633](https://github.com/gordonthree/pca9633)". Each library must have a new name (as defined by the `name` field in library.properties). Please change the `name` value to something that is not already in use and then either bump the library.properties `version` value then make a new tag/release or else redo the existing tag/release.

Once that's done, please comment here so we will know we can proceed.

HendrikVE, 2020-03-19:
Hi @per1234,

thank you for your fast response. I force-pushed my version 1.0.0. I thought the name would be ok, because it was in uppercase instead of lowercase :P

per1234, 2020-03-19:
>I force-pushed my version 1.0.0

I see you changed the `name` value, but the `v1.0.0` tag (which is the repository's only tag) still points to the ref where the `name` was `PCA9633`:
https://github.com/HendrikVE/ArduinoPCA9633/blob/v1.0.0/library.properties#L2

>I thought the name would be ok, because it was in uppercase instead of lowercase

Here's my reference for it being regardless of case:
https://github.com/arduino/Arduino/issues/7887#issuecomment-415033718

I have updated the Library Manager FAQ so it now contains this information.

HendrikVE, 2020-03-19:
Ok, it should be fixed now. It's still `v1.0.0` but the name is `NXP PCA9633` now.

cmaglie, 2020-03-20:
Done.

closed 2020-03-20 by cmaglie