arduino/Arduino #9892: [Library Manager] Remove LoRaLib

jgromes, 2020-03-19:
Hi,

Could you please remove LoRaLib (https://github.com/jgromes/LoRaLib) from library manager? It's no longer being worked on, so I'm starting to phase it out.

Thanks!

cmaglie, 2020-03-20:
I can't find it this library in the library index, so I'm closing this issue as "done".

Feel free to reopen if I'm missing something...

jgromes, 2020-03-21:
That's odd, since library manager knows about it and will let me install it.
[image]
.

cmaglie, 2020-03-23:
Oh now I can see it :man_facepalming:...

Ok, I've removed it for good now, if you can still see it give some time for the index cache to update.
Thanks.

closed 2020-03-23 by cmaglie