arduino/Arduino #9894: Arduino IDE issue

CliffMonk, 2020-03-20:
Hi, I am new to Arduino and programming, and have installed Arduino IDE from the Windows store.
When open IDE I am unable to read the menus at the top, i dont know if this an IDE or Windows issue.

Please Help!!!!

[image]

per1234, 2020-03-20:
I think you're encountering this bug:
https://github.com/arduino/Arduino/issues/7839

The workaround is shown here:
https://github.com/facchinm/Arduino/commit/fb89a670783d8a45db4998d43ea302ac63147793

Instructions for the workaround:

1. Download the standard version of the Arduino IDE, which you can get by clicking the "Windows Installer, for Windows XP and up" or "Windows ZIP file for non admin install" links on the Downloads page: https://www.arduino.cc/en/Main/Software. It is possible to do this with the Microsoft Store app version of the Arduino IDE, but it's a very complicated process. If you really want to do that, you can do a Google search to find instructions for getting write access to the folder containing the app. After that, you can follow the rest of these instructions.
1. Open the folder containing the Arduino IDE installation
1. Open the file arduino.l4j.ini with a text editor
1. Add the following text:
    ```
    -Dawt.useSystemAAFontSettings=on
    ```
1. Save the file.
1. Restart the IDE if it's running.

I'm going to close this as a duplicate of https://github.com/arduino/Arduino/issues/7839 but if it turns out the workaround doesn't solve the issue for you, let me know and I'll re-open the issue.

closed 2020-03-20 by per1234

CliffMonk, 2020-03-20:
Thanks, i'll give it a try

CliffMonk, 2020-03-20:
Worked a treat, Tanks again