arduino/Arduino #9895: Ctrl-shift-U does not work work when editor is focused

matthijskooijman, 2020-03-20:
I just noticed a small bug: When the editor is focused, ctrl-shift-u for "Upload using programmer" does not work. Ctrl-u for normal upload works, and when removing the focus from the editor (e.g. clicking the messages window below the editor) makes ctrl-shift-u work again.

This was using a few weeks old master version, on Linux.

Can anyone confirm this?

per1234, 2020-03-21:
I confirm.

It occurs on my Ubuntu 19.10 machine with all the IDE versions I have installed, including Hourly Build 2020/03/16 09:25 and 1.6.4 (so it wasn't introduced with the switch to RSyntaxTextArea in 1.6.5).

The issue does not occur on my Windows 10 machine.

ubdussamad, 2020-03-27:
Same thing is happening with me too.

OS: Ubuntu 16.04 , with IDE version 1.8.10