arduino/Arduino #9897: Please add FastMap library to library manager

RobTillaart, 2020-03-21:
Please add  https://github.com/RobTillaart/FastMap  
Thanks!

RobTillaart, 2020-03-22:
@per1234 
How long does it take for a lib to be picked up?
If there is something missing do I get a message?

per1234, 2020-03-22:
>How long does it take for a lib to be picked up?

The addition of the library to the Library Manager index is a manual process. We let the requests build up a bit, then cmaglie goes through and adds them all in one go. The last batch was added a couple days ago, so you're coming in at the beginning of the interval. I'd say it will be two weeks at most. cmaglie will close this issue when the library is added.

After your library is in the Library Manager index, all compliant tags will be automatically picked up by the indexer job in a matter of hours from when you make the tag.

>If there is something missing do I get a message?

I did run your repository through my [validation script ](https://github.com/per1234/ino-library-manager-validator)already and it found no problems, so it should be fine. When cmaglie adds the library to the index there will be a definitive check that the tag is compliant with all the [requirements](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) and cmaglie will comment here if any problems are found.

However, this only happens when the library is initially added to the Library Manager index. If any future tags are not compliant, the only way you will know is by that version never showing up in Library Manager. I do have a [bash script that checks compliance](https://github.com/per1234/arduino-ci-script#check_library_manager_compliance-librarypath), which you could run during your CI build if you want to make sure that doesn't happen.

cmaglie, 2020-03-23:
Done.

closed 2020-03-23 by cmaglie