arduino/Arduino #9905: Please add Max44009 library to library manager

RobTillaart, 2020-03-23:
Please add https://github.com/RobTillaart/Max44009

Question: are library names case sensitive?  Note there exists a MAX44009 library.

https://arduino.github.io/arduino-cli/library-specification/  does not mention this explicitly

Thanks!

per1234, 2020-03-23:
The requirement for each library added to the Library Manager index to have a unique `name` value in library.properties is case insensitive. This is documented here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager
>Your library must not have the same library.properties name value (regardless of case) as another library previously added to the Library Manager index.

Fortunately, there is no existing library in the Arduino Library Manager index with the name "Max44009" in any case variation. There is a library named "MAX44009 library", but that doesn't pose any conflict.

RobTillaart, 2020-03-23:
Yes I saw that one. Thanks for the clarifying link.

cmaglie, 2020-03-23:
Done.

closed 2020-03-23 by cmaglie