arduino/Arduino #9909: problem with pow() and any loop

aghfars, 2020-03-23:
hi i found a problem with pow() in loop and this picture explain that.
[image]

PaulStoffregen, 2020-03-23:
You're truncating when you should be rounding.  Add 0.5 before truncating to integer.  Or use round().

aghfars, 2020-03-24:
This problem only exists in loops.
[image]

and i make my own function to fix this and its work.
Why should I use round () and increase the size of the program?
Why do you refuse to fix this problem?

[image]

per1234, 2020-03-25:
@aghfars I think if you run this sketch the situation will be clear:
```c++
void setup()
{
  Serial.begin(9600);
  int i = 4;
  while (i >= 0)
  {
    double POW = pow(10, i);
    Serial.println(POW, 6);
    i--;
  }
}

void loop()
{}
```
The output will look something like this:
```
9999.998046
999.999816
99.999992
10.000000
1.000000
```
[`pow()` returns a `double`](https://www.arduino.cc/reference/en/language/functions/math/pow/). Floating point numbers are not exact. When you convert 9999.998046 to an `int` it is truncated to 9999. This is why you need to use `round()`.

This is not a bug. It is confusing, but this is just the way it works. The result is influenced by the compiler optimizations, which is why you get a different result without the code that doesn't have the `while` loop. Your `Pow` function returns an `int`, so it doesn't introduce the floating point issue.

I think Arduino's documentation does an OK job of explaining this situation. The return type of `pow()` is documented. We mention the the inexact nature of floating point numbers and truncation in the [`float` reference page](https://www.arduino.cc/reference/en/language/variables/data-types/float/), as well as the solutions. I guess it is problematic that if you check the `pow()` reference page, then the `double` reference page, you would not see the information on the `float` reference page.

closed 2020-03-25 by per1234

aghfars, 2020-03-25:
I re-wrote the function, but with double and no problems in the loop.

[image]