arduino/Arduino #9916: [Library Manager] Please add the Iridium SBD Library to the Library Manager

PaulZC, 2020-03-25:
Hi,
Can you please add https://github.com/PaulZC/IridiumSBD to the Library Manager?
I hope all is going OK where you are.
Sincere thanks,
Paul (PaulZC)

per1234, 2020-03-25:
Hi @PaulZC. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The name "Iridium SBD" is already taken by https://github.com/mikalhart/IridiumSBD.

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

PaulZC, 2020-03-25:
Thank you @per1234!
Take care - stay well,
Paul

cmaglie, 2020-03-26:
Done.

closed 2020-03-26 by cmaglie

PaulZC, 2020-03-26:
Thank you Cristian (@cmaglie)!
I hope all is ~OK in Milan.
Take care and stay safe,
Paul