arduino/Arduino #9919: Feature request - Posibility to name and save device configuration options

juancgalvez, 2020-03-25:
I work with several devices (ESP32, ESP8266, ATMEGA168P based, ATMEGA328P based) and every time I need to compile for a different a one, I must select that particular device and manually change all the related options.

Would be good if I can just name and save each device complete configuration and later access it as a menu item in Tools menu so all options are set in a single step.

per1234, 2021-04-06:
This idea was discussed tangentially from https://github.com/arduino/Arduino/issues/6034#issuecomment-562957101

juancgalvez, 2021-04-06:
@per1234 in fact, that is a completely different requirement. Mine is about saving each device complete configuration settings in a single menu option. The one you referenced is about disabling unused boards devices. Read my first post.

per1234, 2021-04-06:
>The one you referenced is about disabling unused boards devices

That's what the issue is about, but the tangential discussion I linked to is about adding a user configurable "favorite boards" menu to the IDE.

juancgalvez, 2021-04-06:
My request is not about favorite boards. Is about being able to save the device configuration options in some place you can recover in a single action (Could be menu option). Every time I change the target device I have reconfigure all the options. Would be good to recover all those configuration options fast, easy and less prone to errors.