arduino/Arduino #9923: If/curly brackets bug

Enigma86d, 2020-03-26:
In a sketch, the following two statements produce different results (they should produce the same results):
1.
  if (true==false) {
    connectToGPS();
  }

2.
  if (true==false) 
    connectToGPS();

The way the sketch behaves is that if I delete the curly brackets, the function is called although it should not get called.
Demonstration: [https://www.youtube.com/watch?v=H8D3J4NOkkQ](url)

Arduino version: 1.8.12
Using minicore with LTO enabled

matthijskooijman, 2020-03-26:
Hm, that looks weird. Are you sure that it is this particular invocation of that function? You're not calling it elsewhere somehow?

Youre sketch seems rather big, did you try if the same thing happens with a smaller sketch (maybe removing library references to make the sketch more self-contained)? That might make it easier to reproduce.

Could you maybe share the generated .elf files compiled with and without the curly braces? Maybe I can see what changes in the compiled code.

Finally: Can you be a bit more specific about your environment? Do you have a link to this "minicore"? What board do you have selected?

cmaglie, 2020-03-26:
A copy of the **complete** sketch may help too...

Enigma86d, 2020-03-26:
Hi,

I did more digging and apparently is a combination of if statement, curly brackets and #define statement. I managed to isolate the problem in a simple arduino Uno sketch (below).
`#define doSOmehting1(); {Serial.println("Error1");}
#define doSOmehting2(); Serial.println("Error2");

void setup() {
  Serial.begin(9600);
}

void loop() {
  if (false)
    doSOmehting1();
  if (false)
    doSOmehting2();
  delay(300);
}`
In the above example both "Error1" and "Error2" are printed. If I add curly brackets right after the if statement then they are not printed.
I would think that my usage of #define is valid, even if I don't see an example on Arduino webpage to support this [https://www.arduino.cc/reference/en/language/structure/further-syntax/define/](https://www.arduino.cc/reference/en/language/structure/further-syntax/define/). I was using #define in this manner fairly frequently before starting to program in Arduino and I didn't had issues, but maybe it's a wrong usage... I'm not sure.

Enigma86d, 2020-03-26:
[sketch_mar26b.txt](https://github.com/arduino/Arduino/files/4387420/sketch_mar26b.txt)

cmaglie, 2020-03-26:
At a first sight it seems that there is an extra `;` in the defines:

```
#define doSOmehting1(); {Serial.println("Error1");}
                      ^
#define doSOmehting2(); Serial.println("Error2");
                      ^
```

this may explain the issue because `doSomething1()` (in my understanding of macros) is transalted to:
 `; {Serial.println("Error1");}`

so in the example above:
```
if (false)
    doSOmehting1();
```
becomes:
```
if (false)
    ; {Serial.println("Error1");};
```
that is equivalent to:
```
if (false)
    ;   <--- this is an empty if block
{Serial.println("Error1");};
```

matthijskooijman, 2020-03-26:
Yup, that is indeed the problem. The ; after the *invocation* of your macro is not replaced along with the macro, but is left there, and the extra ; in your macro definition is added.

A proper way to write this define would be:

    #define doSomehting1() Serial.println("Error1")

If you then write `doSomething1();` it would become `Serial.println("Error1")`

Note that if you want to expand a macro to *multiple* statements, and still support both if with and without curlies, you would typically write:

    #define doSomehting2() do { Serial.print("Error: "); Serial.println(error); } while (0)

See e.g. https://stackoverflow.com/questions/154136/why-use-apparently-meaningless-do-while-and-if-else-statements-in-macros for more info on this trick.

So, I'm pretty sure this is a problem in your code, the compiler is behaving as it should. I'm going to go ahead and close this issue, feel free to post followup comments if needed.

closed 2020-03-26 by matthijskooijman

Enigma86d, 2020-03-26:
Yes, sorry about the false alarm.
Due to the way the define syntax is  (#define CONST_NAME VALUE) I though that the space between CONST_NAME and VALUE dictates where the separation is made between the two, but it seems that it is not how it actually works. Sorry