arduino/Arduino #9924: New library: NMEAParser

deadbird99, 2020-03-26:
Hi.

I'd like to add my library to Arduino Library Manager. It's available here: https://gitlab.com/X99/NMEAParser

Thanks in advance!

per1234, 2020-03-26:
Hi @deadbird99. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The name "NMEAParser" is already taken by https://github.com/Glinnes/NMEAParser

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

per1234, 2020-06-14:
Closing due to lack of response.

@deadbird99 if you ever get around to making your library compliant, please comment here and I'll re-open this issue.

closed 2020-06-14 by per1234