arduino/Arduino #9929: Ev3-arduino communication library

tasos12, 2020-03-27:
Hello
Please add the initial version of my library
https://github.com/tasos12/ev3-arduino

per1234, 2020-03-27:
Hi @tasos12. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

I see you have already [changed to using a compliant `name`](https://github.com/tasos12/ev3-arduino/commit/2cf4f581b930c2f18bc04b08ac86424260eb7251), but you have not made a tag/release since then. The Library Manager indexer works from tags, so your library is still non-compliant as far as the indexer is concerned.
You'll need to update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

tasos12, 2020-03-27:
Thanks for the update.  I made the necessary changes

cmaglie, 2020-03-27:
Done.

closed 2020-03-27 by cmaglie