arduino/Arduino #9934: New Library: CurrentTransformer

dirkx, 2020-03-27:
Library to handle current coils; with measurements and  async call backs when power appears/is lost. Used extensively at https://makerspaceleiden.nl/.

https://github.com/dirkx/CurrentTransformer/releases/tag/1.0.0
https://github.com/dirkx/CurrentTransformer.git

per1234, 2020-03-27:
Hi @dirkx. Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/JChristensen/CurrentTransformer

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

dirkx, 2020-03-27:
Understood & apologies. 

I've renamed everything; https://github.com/dirkx/CurrentTransformerWithCallbacks.git is the new name. 

Also pushed out a major version number increment; as this obviously breaks compatibility with the code we have at the https://makerspaceleiden.nl/ where we use this a lot.

Release is at https://github.com/dirkx/CurrentTransformer/releases/tag/2.0.1

cmaglie, 2020-03-27:
Done.

closed 2020-03-27 by cmaglie