arduino/Arduino #9944: [Library Manager] Please add BLESerial

Uberi, 2020-03-28:
Please add my new library to the library manager:
https://github.com/Uberi/Arduino-BLESerial

Thanks!

per1234, 2020-03-28:
Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value used in your library has already been taken:
https://github.com/5pIO/BLESerial

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

Uberi, 2020-03-29:
Hi @per1234,

Thanks for letting me know! I've renamed the library to HardwareBLESerial, since this library is more about built-in BLE hardware.

per1234, 2020-03-29:
You also need to make a Git tag or GitHub release that contains change to the updated `name`. Your only existing tag still has the name set to BLESerial:
https://github.com/Uberi/Arduino-HardwareBLESerial/blob/1.0.0/library.properties#L1

Uberi, 2020-03-30:
Oops, thanks @per1234 - I've changed the tag to the latest commit now.

cmaglie, 2020-04-01:
Done.

closed 2020-04-01 by cmaglie